Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #2: Need connection between Sim800l DTR pin and ESP32

toomasz, 2019-07-09:
DTR pin from sim800 is used for entering, exiting sleep mode. Is there a connection from it to esp32?
See: 
https://github.com/toomasz/SimcomGsmLib/blob/1e2de6eb4a25372827a03422578f388ed87ab158/src/SimcomAtCommands.cpp#L560

vshymanskyy, 2019-07-09:
I just uploaded the schematic: https://github.com/Xinyuan-LilyGO/TTGO-T-Call/blob/master/datasheet/TTGO_T-Call_SIM800_v1.3_schematic.pdf

Please let us know about your recommended changes to the schematic for future revisions of the board. Thanks!

toomasz, 2019-07-11:
@vshymanskyy So I recommend to connect this DTR pin of SIM800l to any free GPIO of esp32 to allow entering exiting sleep mode

toomasz, 2019-07-31:
Any estimate when next revision will be available?

vshymanskyy, 2019-09-29:
Ok, closing this as it is already covered by this request: https://github.com/Xinyuan-LilyGO/TTGO-T-Call/issues/13

closed 2019-09-29 by vshymanskyy