Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #3: ESP32 Arduino + PPPoS example

vshymanskyy, 2019-07-09:
It would be a very good addition to have a PPPoS example for ESP32.
Waiting for comments in this tickets:
https://github.com/espressif/arduino-esp32/issues/1858

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

levkovigor, 2021-08-25:
> It would be a very good addition to have a PPPoS example for ESP32.
> Waiting for comments in this ticket:
> [espressif/arduino-esp32#1858](https://github.com/espressif/arduino-esp32/issues/1858)

https://github.com/levkovigor/ppposclient

Dirtthaller, 2022-07-13:
> > It would be a very good addition to have a PPPoS example for ESP32.
> > Waiting for comments in this ticket:
> > [espressif/arduino-esp32#1858](https://github.com/espressif/arduino-esp32/issues/1858)
> 
> https://github.com/levkovigor/ppposclient

Unfortunately your lib doesn't even build. I guess it stopped working since a few new releases of framework. Are there any hints on how to make it work nowadays?

vshymanskyy, 2022-07-13:
It needs a complete rework, unfortunately

levkovigor, 2022-07-13:
> Unfortunately your lib doesn't even build. I guess it stopped working since a few new releases of framework. Are there any hints on how to make it work nowadays?

what version of esp32-arduino are you using? Library has been tested with version 2.0.0

Dirtthaller, 2022-07-13:
@levkovigor I tested it on 2.0, 2.0.2 and 2.0.4 board versions in arduino ide. Always the same problem related to pppapi methods
[image]
I also tried in platformio with latest toolchain and framework. Pretty much the same output. Here are the errors I faced.
**.pio/libdeps/esp32dev/PPPOSClient/src/PPPOS.c:43:1: error: unknown type name 'ppp_pcb'**
**.pio/libdeps/esp32dev/PPPOSClient/src/PPPOS.c:203:47: error: 'ppp_output_callback' undeclared (first use in this function); did you mean 'dns_found_callback'?**
**.pio/libdeps/esp32dev/PPPOSClient/src/PPPOS.c:210:30: error: 'PPPAUTHTYPE_PAP' undeclared (first use in this function); did you mean 'IP_SET_TYPE_VAL'?**
I found out that ppp_pcb undeclared and ppp_output_callback can be solved by defining PPP_SUPPORT in the lvip/opt.h
[image]
[image]
I know that normally this define goes from IDF config file.
Getting rid of this
.pio/libdeps/esp32dev/PPPOSClient/src/PPPOS.c:210:30: error: 'PPPAUTHTYPE_PAP' undeclared (first use in this function); did you mean 'IP_SET_TYPE_VAL'?
and this 
.pio/libdeps/esp32dev/PPPOSClient/src/PPPOSClient.cpp:159:59: error: 'fd' was not declared in this scope
righ after defining CONFIG_LWIP_PPP_SUPPORT 1 in lvipopts.h lead to linking problems such as 
[image]

in @vshymanskyy library I faced the same issues except there is no fd() and PPPAUTHTYPE_PAP errors. In general all the behaviour is pretty same: defining CONFIG_LWIP_PPP_SUPPORT to enable PPP leads to linking issues. Maybe you guys know the easy workaround for that? Or mb I did something wrong. Kindly hope for your advice)

vshymanskyy, 2022-07-13:
you have an option to build with pre-2.0 version of Arduino Core

levkovigor, 2022-07-13:
@Dirtthaller I've tested it now, all is ok:

[image]
[image]
[image]

Dirtthaller, 2022-07-13:
@levkovigor I guess I need to make a try with fresh install of all stuff. Thank you for your time! Stay safe!

Dirtthaller, 2022-07-13:
> you have an option to build with pre-2.0 version of Arduino Core

I assume it is the only way for now. Will stick to that approach. Thanks)

alexandreponso, 2023-03-12:
I was able to compile the example and running it with 2.0.7 version on a VERY dirty way:

1) I took the code at https://github.com/espressif/esp-lwip, and put all stuff related to ppp into my project folder (src/netif/ppp folder).
2) Changed the sdkconfig.h file (AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\tools\sdk\esp32\qio_qspi\include) to add the following lines: 
    #define CONFIG_LWIP_PPP_SUPPORT 1
    #define CONFIG_LWIP_PPP_PAP_SUPPORT 1

Thermelgy-Repo, 2023-06-06:
> . I guess it stopped working since a few new releases of framework. Are there any hints on how to make it work nowadays?

Hai @alexandreponso, 
I am also facing the compile issue. I am little afraid to do the copy of the files from the esp-lwip to the ppp folder, It may have unpredicted error. Is it working?. Any other alternative methods?

alexandreponso, 2023-06-12:
Hi Xinyuan,

Copying the files was the only way that I got it working. And worked
surprisingly well. But I stopped working with ESP32, I'm facing MQTT issues
when connecting to several brokers with TLS. Although using PPPoS or Wifi