Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #4: Set sim power pin 23 high

crash007, 2019-07-10:
https://github.com/Xinyuan-LilyGO/TTGO-T-Call/blob/6848ddee1c57b1c8b12c7d8b9594519e28801e0f/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino#L65

Shouldnt pin 23 be set to high aswell?
Best regards

vshymanskyy, 2019-07-11:
I will be re-checking the examples after I get the latest revision of the board.

crash007, 2019-07-11:
Sounds good. I have order som boards. I can help you test if i receive them before you.

crash007, 2019-07-17:
I received my modules today and can confirm. I made a small pull request https://github.com/Xinyuan-LilyGO/TTGO-T-Call/pull/5

vshymanskyy, 2019-07-17:
Thanks. Receiving mine soon. Will test and update. Please check out the GSM OTA example!

vshymanskyy, 2019-07-18:
The TinyGSM and OTA update examples are updated. Please check them out!

closed 2019-07-18 by vshymanskyy