Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #7: redefinition error due to utilites.h in examples

crash007, 2019-07-19:
I cant compile the examples due to redfinition error in utilites.h
Suggested fix would be:
```
#ifndef UTILITIES_H_INCLUDED
#define UTILITIES_H_INCLUDED
...
#ENDIF
```

Error log:
`./utilities.h: In function 'bool setPowerBoostKeepOn(int)':
../utilities.h:6:6: error: redefinition of 'bool setPowerBoostKeepOn(int)'
 bool setPowerBoostKeepOn(int en)
      ^
In file included from ../sloeber.ino.cpp:25:0:
../utilities.h:6:6: note: 'bool setPowerBoostKeepOn(int)' previously defined here
 bool setPowerBoostKeepOn(int en)`

vshymanskyy, 2019-07-26:
By the way, do you build with Arduino IDE or Platformio?

crash007, 2019-07-27:
Im using Eclipse.

emigbur, 2019-09-05:
Hi, I got this error
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "ESP32 Wrover Module, Default, QIO, 80MHz, 921600, None"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\emigbur\Documents\ArduinoData\packages -hardware C:\Users\emigbur\Documents\Arduino\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\emigbur\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\emigbur\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\emigbur\AppData\Local\Temp\arduino_build_650087 -warnings=default -build-cache C:\Users\emigbur\AppData\Local\Temp\arduino_cache_902053 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.esptool-2.3.1.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\emigbur\Documents\Arduino\TTGO-T-Call_v1\TTGO-T-Call_v1.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\emigbur\Documents\ArduinoData\packages -hardware C:\Users\emigbur\Documents\Arduino\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\emigbur\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\emigbur\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32wrover:PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\emigbur\AppData\Local\Temp\arduino_build_650087 -warnings=default -build-cache C:\Users\emigbur\AppData\Local\Temp\arduino_cache_902053 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.esptool-2.3.1.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\emigbur\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\emigbur\Documents\Arduino\TTGO-T-Call_v1\TTGO-T-Call_v1.ino
Using board 'esp32wrover' from platform in folder: C:\Users\emigbur\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0
Using core 'esp32' from platform in folder: C:\Users\emigbur\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/config" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid/api" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_trace" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_update" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bootloader_support" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bt" "-IC:\\Users\\emigbur\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.0/t [... cut ...] \AppData\\Local\\Temp\\arduino_build_650087\\sketch\\TTGO-T-Call_v1.ino.cpp" -o nul
TTGO-T-Call_v1:41:23: error: utilities.h: No such file or directory

compilation terminated.

Using library Wire at version 1.0 in folder: C:\Users\emigbur\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0\libraries\Wire 
Using library TinyGSM at version 0.7.9 in folder: C:\Users\emigbur\Documents\Arduino\libraries\TinyGSM 
exit status 1
utilities.h: No such file or directory

How can I install this file?

closed 2021-01-19 by lewisxhe