Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #8: TinyGSM example wont start from battery

crash007, 2019-07-19:
If I power the module from battery nothing happens. If I start with power from usb and then unplug usb it will continue to run from battery power.
maybe something with Serial or power?

My battery had 4.1V when doing the test.

crash007, 2019-07-19:
I tried to set pin 23 to high first in setup(). Measureing pin 23 shows its 0v

crash007, 2019-07-20:
I dont think the tinyGSM example continues to run on battery, it only continues to blink a few times before stopping.
So i tested with my own mqtt example. If i pull the usb plug with battery connected it will contine to blink a few times and then it stops.

I found the following in documentation for ip5306 which i tried with no success

> Register address=0x23
> Bit(s) Name Description R/W Reset
> 7:6 Reserved
> 5 charging constant current loop selection:
> 1: VIN terminal CC constant current
> 0: BAT terminal CC constant current
> RW 1
> 4:0 Reserved RW

So I made this method:
```

#define IP5306_Charger_CTL1  0x23

bool setBatteryIn(){
	Wire.beginTransmission(IP5306_ADDR);
	  Wire.write(IP5306_Charger_CTL1); //0x23
	    Wire.write(0x00); //bit5 0 = battery
	  return Wire.endTransmission() == 0;
}

```

crash007, 2019-07-20:
Key pin on IP5306-I2C is connected to Reset button with a pull up from VCC3V3.
BUT VCC3V3 is 0v due to IP5306-I2C not activated.
Solution is to connect Battery to VCC3V3 and press Reset-button for 1 Second.
Also i dont use setBoost method.

Reference:

> Translated section of IP5306 datasheet: "IP5306 can recognize long key and short key operation, PIN5 pin is vacant when the key is not needed. ● The button lasts longer than 50ms, but less than 2s, it means the short press, short press will turn on the power indicator and boost output. ● If the duration of the key is longer than 2s, the long press will activate the light and turn it on or off. ● less than 50ms button action will not have any response. ● Pressing the short key twice within 1 second will turn off the boost output, battery level indicator and lighting LED"

crash007, 2019-07-21:
So now i found out whats wrong. forget everything i wrote earlier.
Brownout detector was triggered 
So I put a capacitor 470uF between GDN and 5v and now it works good.

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

crash007, 2019-07-23:
So i tried with another module and it works fine from battery. Maybe i was pressing longer than 2 seconds when booting up which will disable boost.

For referense datasheet: https://docviewer.yandex.ru/view/0/?*=GWwk9TU8V4qspLWekIykObr8keV7InVybCI6InlhLWRpc2stcHVibGljOi8vZTlnMElGZVlCM3ZwaTlxbklETy9wTThHZzRXOVhCc1F4UmRqNHZqYzNxVEJtay9COU1EVkFpV1BUTS9wVFR6QnEvSjZicG1SeU9Kb25UM1ZvWG5EYWc9PSIsInRpdGxlIjoiSVA1MzA2IEkyQyByZWdpc3RlcnMucGRmIiwibm9pZnJhbWUiOmZhbHNlLCJ1aWQiOiIwIiwidHMiOjE1NjM4NDU2MzkwNTQsInl1IjoiNjU1Mzk1ODg5MTU1ODI2OTAzMSJ9

https://github.com/m5stack/M5Stack/blob/master/src/utility/Power.cpp

closed 2019-07-23 by crash007

jbrepogmailcom, 2020-01-24:
Hello does it mean that the module starts from battery right away? Do you use boost? What are your IP5306 settings please?

ClemensGruber, 2020-01-26:
Would be interested also! I have to press the reset button still!

ClemensGruber, 2020-02-06:
Is there any workaround for auto-starting the T-Call after connecting a battery? Due to the high power consumption in deep sleep I thought that a ultra low power timer like the TPL5110 can switch the T-Call. But for this scenario it has to wake up after switching battery connection without pressing any reset / button!

jbrepogmailcom, 2020-02-06:
Tried many ways, no success. See my issue #40

ClemensGruber, 2020-02-06:
Beside this problem there may be more with the IP5306 after sleep phases > 30 s:
https://github.com/m5stack/M5Stack/issues/156#issuecomment-487809103