Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #10: SPI + SIM800

fgerrad, 2019-07-26:
Hi. How to use SIM800 and SPI at the same time when pins 23 (MOSI) and 5 (SS) are used by SIM800?

ianwillianb, 2019-08-09:
You can redefine the SPI pins to any other free GPIO.
Call the method  SPI.begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1);

closed 2019-08-14 by vshymanskyy

Electra99, 2019-11-10:
There are three SPI interface on ESP32-D0WD chip into the ESP32 WROVER-B module: SPI, HSPI and VSPI.
HSPI could be used ;-) .....but notice that GPIO13 (HSPI MOSI) is also connected with the Blue LED. 
VSPI could theoretical been used by using "SPI.begin...", but NOT recomended because you can't use any standard software library, when the GPIO4 (SIM800 PWKEY) , GPIO22 (I2C to Power IC IP5306) and GPIO23 (SIM800 PowerOn) pins fro ESP32-D0WD are already been used into TTGO T-CALL board. For the SPI interface not any pin are availible on the TTGO T-CALL board. I have test to connect a 3.5" TFT display (with touch screen) with HSPI to TTGO T-CALL and it works great :-)

uglagwox, 2020-07-31:
@Electra99 
Hi, could you please describe in detail how you connected your 3.5" TFT over SPI, please?
I've tried 3-4 (or even more) SPI pin combinations with a 2.4" TFT with ILI9341, but the screen is always just plain bright white (only LED light).
The TFT is definitely OK as it works just fine over SPI on a NodeMCU (Lolin v3.0) .
Have tested with Micropython (using two different ILI9341 drivers) and also with Arduino example (the same lib as for NodeMCU, but of course with pins defined for ESP32).

@lewisxhe 
Maybe you have an idea what's wrong? (have also tried using the SPI pins as in your comment regading SD card over SPI)

Thanks for any help in advance!

kousiksasmal, 2021-05-13:
@Electra99 
Hello, could you please share the code or the library which you used for HSPI communication.

Thanks a lot.