Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #11: When battery powered,  will not work if battery bellow ~4V

nikil511, 2019-08-03:
When T-Call is powered by battery (with TX/RX connected via serial UART not USB) if battery voltage drops bellow ~4V I get brownout errors.

I am powering the device with multiple 18650 in parallel, so there are enough Amps.

crash007, 2019-08-04:
Had the same problem. Activate keep boost on solved it for me. Check the examples.
also you can check my closed issue.
Good luck

nikil511, 2019-08-04:
Are you talking about issue 8 ? I think what I am describing is different, more complex scenario. 
I am activating BatteryBoost and I can go bellow 4V if I start with USB and I remove the USB.
Also if I start with a +4V battery  which eventually drops bellow 4, thats not a problem either.

The problem is that once you are bellow 4V, with BatteryBoost on, No USB, and ...**you press reset**, then IP5306 BatteryBoost is disabled again, then ESP32 can not start in order to enable BatteryBoost again, so device is stuck in brownout error loop.

The solutions I see are:
 a) IP5306 reset should not change its values back to default. Is there a permanent "Save" ?
b) T-CALL reset button, should not reset IP5306 but only ESP32. This requires hardware modification.

crash007, 2019-08-05:
I did put a 470uF capacitor between 5v and GND but that is not really necessary if you get the boost to work. when i manage to enable boost I could remove the capacitor.

nikil511, 2019-08-05:
it seems I am repeating yours steps :-)
I also tried a CAP at 5V with border line success (not stable)
I also found out I was working on a dodgy module that had 4.5V on 5V thus was more sensitive. My other module had 5V indeed.
 
So for enabling boost I was using code from the[ M5STACK power class](https://github.com/m5stack/M5Stack/blob/master/src/utility/Power.cpp) and the example from @vshymanskyy 

However I have not tampered with CTL1 & CTL2 yet
Thanks for the tip! I will be back shortly :-)

nikil511, 2019-08-05:
Problem persists.
I am trying on 4 different modules, (two of them have a proper 5V line)
I set the registers first thing in setup.
If I remove USB everything works fine with battery at 3.7V
If I press reset, it boots normally, executes the set registers part but as soon as it tries to turn on the SY8089 4v4 regulator for SIM800 (MODEM_POWER_ON set High) it gets a brownout error.

I know that PowerBoost works, because if I turn it off, then it will not reach that far! It crashes immediately upon reset/boot. With PowerBoost ON the ESP32 works fine, its only when SIM800 gets activated that it fails with brownout.

@crash007 Have you tried resetting with a low bat AND a firmware that is using GSM ?

nikil511, 2019-08-06:
ok, I have answers.
 
1 unit does not work reliably (brownout / guru meditation /etc)  with battery no mater what the voltage! Addition of capacitor made things better but not good enough.

2 units work ok up to ~4V. When I added capacitors AND replaced the thin cable of the tiny plug, they can reach 3.3v on battery without problem 

1 unit works up to 3.3v without having to add capacitor or replacing plug cable.

So thats 1/4 for reliability :-(
However @crash007 your registers do indeed fix the problem if the hardware is not messed up.

closed 2019-08-06 by nikil511

crash007, 2019-08-20:
I have been monitoring my vcc and now when the battery has droped below 4v it doesnt upload any more.  @nikil511

nikil511, 2019-08-22:
So @crash007 you confirm my findings, and it's not just a matter of I2C SY8089 settings. 
@lewisxhe @vshymanskyy  you need to increase capacitor at 5V or 4V vbus for the GSM to work on battery. Please see #13

crash007, 2019-08-22:
Yes I can confirm your findings. I have monitored the vcc and when i get
below 4v it stops to transmitt messages.

imranaalam, 2019-11-28:
@nikil511 i am facing the same  problem. what is " thin cable of the tiny plug" ; can you please attach the picture? 
any more insights since last time?
have you tried some other solutions other than ttgo-call? ESP32 + sim800l?

nikil511, 2019-11-29:
I have tried many combinations of ESP32 with some external gsm module.
Problem is that are all prone to noise on the tx/rx thus an all in one
board is better.

Thin cable issue: i have noticed instability problems (brownout) when
powered by the included battery cable and one 18650 not fully charged. When
i skipped that cable and plug and powered tcall directly with a soldered
thicker cable, added capacitor on 5v, those problems went away and tcall
was working with battery at 3v.

Thus I conclude that the combination of battery cable, plug and capacitors
on tcall are not suitable for the 2A gsm spikes, and ideally all three need
to change to be on the safe side and allow operation with empty batteries.

RamiRihawi, 2019-12-01:
@nikil511 
I have the same issue where TTGo-Tcall is not running from battery not even if the battery is above 4v.
it shows the same symptoms where running it from USB works fine, and removing the USB mid way while connected to battery works as expected, but once the ESP is reset i face the brownout loop issue.

I even tried to connect the battery pins directly to DC power supply and set the latest to 4.1V same issue.
it seems that the module can only be powered from the USB port.

P.S. i tried the same methods on two different modules, both bought from Bangood. the two behave exactly the same.

1- When you mentioned adding Capacitors, where exactly did you add them. To the raw of Capacitors connected to the sim module 5v rail from the booster IC? (adding more capacitors to the 3 capacitors delivering the boosted 5V)?

2- have  you tried to change the aforementioned capacitors to maybe better quality ones?

any other tips based on the tests i've done already?

nikil511, 2019-12-01:
You need to enable the IP5306 boost, over I2C. Have a look at the example
code. Capacitor is needed for using GSM at low voltage. You connect it at
5v and GND pins.

kabukimandelux, 2020-01-14:
Had the same issue, thanks for the suggestion with the 470uF capacitor. Seems to work fine now!
Wired to 5v pin and ground

jbrepogmailcom, 2020-01-21:
Dear @nikil511 Manolis, seems that you are very experienced with battery optimization and you know most of all other people. I also had problem with thin wires, so I connected LiPo battery directly to 5V rail, but it is still shutting down when under 4V.

I think it would help many other people - can I please ask you to write short howto, how to optimize the module for GSM battery use? Like 1) use thick wires connected directly to 5V, 2) use capacitor, 3) use library for BatteryBoost... Especially I am not clear how to use the battery boost, if it has to be enabled at each start, or can be activated through some registers or wiring... I also guess that many of us use wakeup from sleep by interrupt from sensor - is that approach compatible with batteryboost?

Thank you very much for your kindness, Jan

nikil511, 2020-01-21:
Have you tried the tinygsm example?
https://github.com/Xinyuan-LilyGO/TTGO-T-Call/tree/master/examples/Arduino_TinyGSM

it contains the batteryboost part and should work with battery + better
cables + capacitor at 5V
Wake from interrupt is irrelevant to IP5306, so will work.

I will tideup my repo and publish examples in the future, but here are the
important registers of the IP5306 and what they do based on my research.
My goal was to somehow make the system wake up after power failure.
Apparently that is not possible, reset has to be pressed. To make things
worst bellow 3V IP5306 enters a zombie mode where it consumes 50mA without
powering the rest of the board, thus a non-protected lithium battery will
go down to zero (!)

#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00
#define IP5306_REG_SYS_CTL1 0x01
Wire.beginTransmission(IP5306_ADDR);
Wire.write(IP5306_REG_SYS_CTL0);
Wire.write(0b110111);
// 0x37 = 0b110111 TCALL example
/*
[1] Boost EN (default 1) [EXM note: if 0 ESP32 will not boot from battery]
[1] Charger EN (1) [EXM note: did not observe difference]
[1] Reserved (1) [EXM note: did not observe difference]
[1] Insert load auto power EN (1) [EXM note: did not observe difference]
[1] BOOST output normally open ( 1) [EXM note: if 0 will shutdown when
ESP32 sleeps after 32s]
[1] Key off EN: (0) [EXM note: could not detect difference]
*/
Wire.write(IP5306_REG_SYS_CTL1);
Wire.write(0x1D); // Set HEX:1D DEC:29 BIN:11101
/*
[1] Turn off boost control signal selection: short press twice
[1] Switch WLED flashlight control signal selection: short press twice
[1] Short press switch boost: disabled
[0] Whether to turn on Boost after VIN is pulled out: opened
[1] Batlow 3.0V Low Power Shutdown EN: enabled
*/

jbrepogmailcom, 2020-01-21:
I see, thanks. I took some different approach from elsewhere, but I have no idea if it works. Maybe I should connect battery back to VBAT and not to +5V, as I had until now (I realized I was bypassing IP5306). But I think VBAT connector is too bat it did not work for SIM800L..

// I2C for SIM800 (to keep it running when powered from battery)
TwoWire I2CPower = TwoWire(0);

#define IP5306_ADDR          0x75
#define IP5306_REG_SYS_CTL0  0x00

bool setPowerBoostKeepOn(int en){
  I2CPower.beginTransmission(IP5306_ADDR);
  I2CPower.write(IP5306_REG_SYS_CTL0);
  if (en) {
    I2CPower.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
  } else {
    I2CPower.write(0x35); // 0x37 is default reg value
  }
  return I2CPower.endTransmission() == 0;
}

nikil511, 2020-01-21:
You can bypass ip5306 and all problems related to it, by providing 5v to
the 5v line using an external step up (eg lipo rider)

But putting a battery (3-4v) to the 5v is not a long term solution :-)

jbrepogmailcom, 2020-01-21:
May I ask for one more advice? When using the code you suggested, it looks like the ip5306 shuts down the power rail and esp32 after those 32 sec when the module is on battery and goes to sleep. Which shuts off also connected sensors, so I can not wake it back by interrupt. Reset is the only way.

schmeckm, 2020-01-31:
Hi Nikil
you have mentioned that you can boot the Board directly over 5 V Line. 
I checked the Pinout of the Board, but I can not see a "VIN" to connect a Lipo rider?
What kind of Pin are you using as a Input?
Markus

nikil511, 2020-01-31:
Pin "5V" :-)

schmeckm, 2020-01-31:
Ok its strange if I connect the Board to the USB than I get 5 Volt out.
I was not aware of it that I can also connect this 5 Volt Pin with a external Power Supply.

Markus

HuxleyFred, 2020-11-29:
Hi, just a comment with respect to the example code above - specifically:

if (en) {
I2CPower.write(0x37); // Set bit1: 1 enable 0 disable boost keep on
} else {
I2CPower.write(0x35); // 0x37 is default reg value
}

If you look at the data sheet (unfortunately written in Chinese), then it indicates that these registers contain reserved fields that should be maintained when updating them.  So you should use a read-modify-write sequence to change them.  i.e. if you wish to just set the two 'boost' related bits you should first read this control register, AND in 0x22 and then write it rather than just writing 'magic numbers'.

See:
https://github.com/m5stack/M5-Schematic/blob/master/Core/IIC_IP5306_REG_V1.4.pdf

bhupiister, 2021-01-26:
@nikil511 if you are still active on this thread @lewisxhe if you can help on this. 
I have removed the battery plug on back and soldered the thick wire directly on board. I am facing issue of board consuming 50mA current when I try to power the board on battery. However, Once I press reset and it enters sleep mode, it consumes 1-2mA.
I have also attached 450uF capacitor between 5v and GND. I just need a way of avoiding 50mA current if the board resets accidentally on battery.

nikil511, 2021-01-26:
Tcall has 3 versions with different power management circuit. If you are
taking about 1st version, battery has to stay connected and above 3v else
reset has to be pressed.

I remember the 50ma problem you mention, it's a zombie mode when the ip5306
is not properly initialized.

If you want to use it in unattended setup, with unstable power, (eg battery
+ solar) then skip its ip5306, and feed it 5v directly in to the bus pin
using an external 5v power management module.
...or just buy a new tcall. Have not tried latest version, but i assume
they have fixed it (else way all these versions?)

bhupiister, 2021-01-26:
@nikil511 can I power it via 3.3 V on 3v3 and GND pin? I am ordering TP4056 with LDO to regulate output at 3.3 volt with capacitors to surpress spikes if any. Will this give also generate 5v via onboard registers?

bhupiister, 2021-01-26:
@nikil511 I am using version with ip5306 ,2019 version.

nikil511, 2021-01-26:
I have not used the latest version, but 5v has to exist, as another
regulator makes 4v for gsm and 3v for esp32.

kekko7072, 2022-01-31:
I am thing to connect a _3.7 V 2000mAh Lipo battery_ to the battery pin but when I connect it says 
```
Brownout detector was triggered
11:13:20.243 -> 
11:13:20.243 -> ets Jun  8 2016 00:22:57
11:13:20.243 -> 
11:13:20.243 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT	ets Jun  8 2016 00:22:57
11:13:20.484 -> 
11:13:20.484 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:13:20.484 -> configsip: 0, SPIWP:0xee
11:13:20.484 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:13:20.484 -> mode:DIO, clock div:1
11:13:20.518 -> load:0x3fff0018,len:4
11:13:20.518 -> load:0x3fff001c,len:1216
11:13:20.518 -> ho 0 tail 12 room 4
11:13:20.518 -> load:0x40078000,len:10944
11:13:20.518 -> load:0x40080400,len:6388
11:13:20.518 -> entry 0x400806b4

```

This is the port 
[image]

This is the battery
[image]

**What battery should I use to make it work using the battery?**

GyTe, 2024-01-02:
Hopefully this will help someone in my same situation. I've find out a "solution" merging information from here and outside. At least for me I been able to power from a solar panel (standard 6 volt - 2 watt) a Lilygo T-Call Sim800L (hardware revision 20190610 - IP3506) using a TP4056 charging module (the revision that accept usb-c as input)

Basically I've connected the solar panel to TP4056 Input, a 18650 to battery (to store energy) and output of TP4056 to the battery connector of Lilygo. Then I put a capacitor on 5 volt (as per above suggestion). 

The first start need to be done pressing the reset button but at least for now (1 month on the roof) everything seems to be ok. My use for now is to retrieve temperature / humidity and battery voltage, post on Thingspeak and then deepsleep for 1 hour. 

To have everything in the same place and in order to avoid rain / humidity ecc I've put everything inside a box used for electric junction. I did just a little hole for the BME280 sensor (in order to pick-up external temperature) and another hole to use external GSM antenna. 

I'm quite sure there're some other "elegant" solution and I'm quite sure the battery life can't run forever (hot and cold weather do not help) but at least for now I'm happy of the result. 

For now I'll leave the box on my roof, then when I'm sure that everything is fine, I'll replace the battery with a brand new, charge with external power supply and then travel to a remote location where I need to pick-up that information (more or less 50km far from my home).  

Hope this help and thanks to everybody for the suggestion.

P.S. In case battery will discharge completely and solar panel will be enough to charge it again, as you know, the Lilygo wont power-up automatically again but honestly I don't have the knowledge to fix that.... as per now everything will run as per above setup.

ridwanmullick, 2024-03-12:
**nothing worked but the settings and configurations of this code seems to be working**

/*
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-sim800l-publish-data-to-cloud/
  
  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files.
  
  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*/

// Your GPRS credentials (leave empty, if not needed)
const char apn[]      = ""; // APN (example: internet.vodafone.pt) use https://wiki.apnchanger.org
const char gprsUser[] = ""; // GPRS User
const char gprsPass[] = ""; // GPRS Password

// SIM card PIN (leave empty, if not defined)
const char simPIN[]   = ""; 

// Server details
// The server variable can be just a domain name or it can have a subdomain. It depends on the service you are using
const char server[] = "example.com"; // domain name: example.com, maker.ifttt.com, etc
const char resource[] = "/post-data.php";         // resource path, for example: /post-data.php
const int  port = 80;                             // server port number

// Keep this API Key value to be compatible with the PHP code provided in the project page. 
// If you change the apiKeyValue value, the PHP file /post-data.php also needs to have the same key 
String apiKeyValue = "tPmAT5Ab3j7F9";

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22
// BME280 pins
#define I2C_SDA_2            18
#define I2C_SCL_2            19

// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

// Define the serial console for debug prints, if needed
//#define DUMP_AT_COMMANDS

#include <Wire.h>
#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

// I2C for SIM800 (to keep it running when powered from battery)
[... 107 lines cut ...]
      client.println();
      client.println(httpRequestData);

      unsigned long timeout = millis();
      while (client.connected() && millis() - timeout < 10000L) {
        // Print available data (HTTP response from server)
        while (client.available()) {
          char c = client.read();
          SerialMon.print(c);
          timeout = millis();
        }
      }
      SerialMon.println();
    
      // Close client and disconnect
      client.stop();
      SerialMon.println(F("Server disconnected"));
      modem.gprsDisconnect();
      SerialMon.println(F("GPRS disconnected"));
    }
  }
  // Put ESP32 into deep sleep mode (with timer wake up)
  esp_deep_sleep_start();
} 
[image]