Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #12: If battery is removed and replaced again, a reset press is required for the IP5306 to start

nikil511, 2019-08-03:
I'm powering T-CALL with batteries which are solar charged. In the event of low bat (bellow 2.5V) IP5306 shuts down. Thats normal. However if battery eventually goes above 2.5V it will not automaticly wake up, RESET has to be pressed. 

Is there a register in IP5306 that controls this?

Andre0407, 2019-08-17:
I will powering this modul with a 3.7 V LiPo batterie (18650). Is the energy enough for everything? Also for the SIM800L module?

If i powering it with USB i get the SMS, when i switched to batterie it doesn't.

eiriksels, 2019-11-27:
@nikil511 I have the same issue. Do you know if there is a specific threshold for when the IP5306 shuts down? My thinking is that if there is I could implement some code that has the board go to "sleep" when e.g. 0.4 v above this threshold. It could then wake up like every hour and check if battery level has changed and go back to normal operation if so.

nikil511, 2019-11-27:
Yes, that's exactly what I also do.

At 3v things are unstable for esp32 anyway.
I have managed to go up to 2.7v using a Capacitor on the 5v bus with
everything working (gsm too)

So my threshold recommendation is something above 3v

eiriksels, 2019-11-27:
Thanks @nikil511! Any code examples you have working including sleep library to reduce power usage when waiting for charge would also be appreciated as I have not yet implemented this functionality 👍

jbrepogmailcom, 2020-01-24:
Hello I am trying to get as low power as possible 5mA which T-call is getting in sleep mode is not enough for me. So I power whole board down via relay connected between the board and battery. However, when I turn the power on I have same problem - reset needs to be pressed. The EN pin is not very reliable, it cannot be used.

Does anyone has other solution how to avoid pressing reset than use sleep? I can not use sleep because the board is completely powered off. Thanks

peteakalad, 2020-06-02:
Is this issue still present for the AX192 versions?

bhupiister, 2021-01-13:
> Yes, that's exactly what I also do. At 3v things are unstable for esp32 anyway. I have managed to go up to 2.7v using a Capacitor on the 5v bus with everything working (gsm too) So my threshold recommendation is something above 3v
> […](#)

Any code that you can share for the same ?

closed 2021-01-19 by lewisxhe