Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #13: Suggestion for next revision

nikil511, 2019-08-06:
Dear TTGO team,
Here is a summary of my wishlist :-)

- In PCB v1.3 since you have spare pins, re-purpose one for VBAT. Current plug is too small for SIM800 power requirements and the cable too thin. (Also the pinout image is wrong 5V is not BAT.)

- Consider replacing your battery plug with standard "JST PH 2-Pin" that everybody are using (!). 

- As @toomasz suggested, connect DTR to a spare GPIO  to enable SIM8000 wake/sleep functionality for potential power saving.
 
- Change IP5306 "KEY"circuit so that it auto-starts upon VBAT input. If voltage goes very low IP5306 will shutdown and needs a reset button press to wake up, which is not good for unattended deployments.
 
- Add more/larger capacitors in the 5V line, to improve stability when SY8089 is turned on

Once your apply the above, you've made a great product :-)l

slavino, 2019-08-06:
Let me add my two cents.

The SIM808 (or other GPS/GNSS capable chip) could add positioning functionality and adding SD card slot could make it even more usable out of the box.

nikil511, 2019-08-07:
Or even better SIM7000 :-)

toomasz, 2019-08-08:
SIM7000 would be awesome

Skipper-is, 2019-08-15:
+1 for SIM7000, 
With 2G being switched off around the world, LTE would be a big bonus!

lewisxhe, 2019-08-23:
@nikil511 Ok, thank you for your suggestion, we will fix it in the next version, SIM7000 board we will try, if you have any requirements on the sim7000 hardware you can ask us.

slavino, 2019-08-23:
> @nikil511 Ok, thank you for your suggestion, we will fix it in the next version, SIM7000 board we will try, if you have any requirements on the sim7000 hardware you can ask us.

* External GPS antenna connector (just as LilyGO TTGO T-Beam has)
* External GSM/LTE antenna connector as option to on board antenna
* SD card based either on direct ESP32 interaction (or ATMEGA328P OpenLog variant via SW Serial)
* maybe also charging circuit bypass option (pin or switch) if multiple batteries connected to the board (2 or more 18650 Li-Ion for example - in external holder and external charging circuit that can utilise the type C connector on T-Call board as pass-thru option). Or if user decides to use QI charger (example: https://www.banggood.com/Bakeey-Type-C-Qi-Wireless-Charger-Receiver-Charging-Adapter-For-Oneplus-5-5t-Xiaomi-6-Mi-A1-S8-p-1247353.html?rmmds=buy&cur_warehouse=CN) and seals the box to IPX8 standard he may utilise it as well.

emigbur, 2019-08-23:
HI,
Very good idea, your product, I already buy one, my idea was to be used with solar panel, can you please  provide an easy way step by step, to make the fix on actual versión ?
Regards.

vshymanskyy, 2019-09-29:
SIM7000 version is on the way! ;)

nikil511, 2019-09-29:
Great news!

Please do not use IP5306 again!

When battery is empty it shutdown to protect it but when battery is charged
(via solar) it requires a button key press to start. It is unusable for
unattended deployment.

Check AXP192 seems more sofisticated and has multiple regulator outputs.

Also please consider adding a 5v solar compatible input.

nikil511, 2019-09-29:
Also if you add an external RTC, like tfox, it would be very useful. And
maybe you could connect the RTC alarm to IP5306 KEY (but needs to go from
low to high)

Or maybe take advantage of the sim7000 RTC and battery monitoring by
connecting it directly to the battery with an optional jumper.

eueduardo3, 2019-09-29:
Better than the SIM7000G would be to use the SIM7600SA, so yes it would be awesome.  and yes the idea of ​​placing an SD card slot RTC would also be sensational

crash007, 2019-09-29:
5v solar compatible input would be great.

Krezalis, 2019-09-30:
Place for max485 transceiver mapped to usart2

Krezalis, 2019-09-30:
Place for sim chip (VQFN-8)

Krezalis, 2019-09-30:
Holes for i2c oled 128*64 (VCC, GND, SCL, SDA), pads for pull-up resistors

lewisxhe, 2019-10-26:
Please click here to learn about the new version. 
 https://github.com/Xinyuan-LilyGO/T-SIM/issues/1#issue-512775074 
@nikil511 @vshymanskyy @slavino @toomasz @Krezalis

Zoltant7, 2019-12-05:
Hi ! One question the 5V pin is the output from the USB or it's an input 5v pin ??
I don't have micro USB C , so i need a workaround, external power, and i suppose to use the rx tx thru the FT232 ! Thanks

nikil511, 2019-12-05:
Yes, you can give 5v to that pin tcall will work fine. Use tx rx and gpio0
for flashing.

Zoltant7, 2019-12-05:
Thanks for the answer my friend! I couldn't find any scheme to it anywhere!

Gary-cat, 2019-12-06:
Let me know when you get this version working I will buy a lot of units for my remote install applications

Zoltant7, 2019-12-06:
Yes, it's really frustrating when the battery runs out of juice, this would require some sort of power on reset device when the charging reaches some safe levels again let's say above 3.3v somewhere around 3.5v. I'm using right now with plain esp32 an 5v solar panel(in reality the voltage is above 6~8 V on sunny day) so a noisy buck boost converter, after that a  linear step down converter, from that to tpu4xxx charger module. (this is a must because the tpu can handle voltages only to 5.5V otherwise it kicks the bucket)
This was a fire and forget project so the part numbers could be wrong. I will look for them if anyone is interested. ✌

denxhun, 2020-01-01:
This HW desing is really awesome I like it so much. What could be good for the next revision: to support SMS (and possible call) receiving from SIM800 a link between the GSM module RI (Ring In) pin and an inturrupt enabled digital input pin of ESP32 would be really beneficial.

Sample code:
https://github.com/mayurharge/TinyGSM/blob/master/examples/Read_SMS_message_with_interrupt/Read_SMS_message_with_interrupt.ino

Duncro, 2020-01-20:
Fix the mic and speaker pins to work. A plain Sim800l works fine, but this module has nothing coming out the speaker and the mic is extremely soft, using the same mic and speaker as on a standalone Sim800l module.

toomasz, 2020-01-23:
When new version of ttgo t-call will be available on aliexpress?

bindi, 2020-02-28:
Will there be a new revision of the t-call with the RI (Ring indicator) pin available? This is a must-have for low power applications.

closed 2020-09-18 by lewisxhe