Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #14: Arduino ESP32 model, crash

fenyvesi, 2019-08-07:
After compiling my program ( Arduino V1.8.5, ESP32 V1.0.2) the T-Call module crashes (LoadProhibited) in the boot process. I suggest that the reason is the not properly chosen ESP32 module from the Arduino possibilities. I tried with "ESP32 Dev module".

Which module should I select?

Thank you.

The result:

> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1100
> load:0x40078000,len:9232
> load:0x40080400,len:6400
> entry 0x400806a8
> write config...
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> Core 1 register dump:
> PC      : 0x4000c2e0  PS      : 0x00060d30  A0      : 0x800d3075  A1      : 0x3ffb1e20  
> A2      : 0x3ffc1398  A3      : 0x00000000  A4      : 0x000000e0  A5      : 0x3ffc1398  
> A6      : 0xffffffff  A7      : 0x0000000e  A8      : 0x00000000  A9      : 0x3ffb1e00  
> A10     : 0x00000002  A11     : 0x3f4041ce  A12     : 0x0000000f  A13     : 0x0000ff00  
> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000000d  
> 
> Backtrace: 0x4000c2e0:0x3ffb1e20 0x400d3072:0x3ffb1e30 0x400d5477:0x3ffb1e60 0x400ea087:0x3ffb1fb0 0x40088f1d:0x3ffb1fd0
>

crash007, 2019-08-07:
I use Arduino 1.8.9, esp32 v1.0.2, Esp32 dev module. 
What is your code doing?
Are you reading or writing to invalid memory?

fenyvesi, 2019-08-07:
You are right. The crash is caused by not the booting but the first routine that tries to write the memory. I am checking it.
Thank you for your help.

fenyvesi, 2019-08-07:
Yeah, I was stupid.

closed 2019-08-07 by fenyvesi