Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #17: SD card problem

robottini, 2019-08-31:
I'm trying to use an sd card module with the TTGO T-call. I can't use the SPI pins because I need to use also the SIM800L module. The SIM800L uses the standard pins 23 (MOSI) and 5 (CS). I tried to remap the SD card pins using this library: https://github.com/nhatuan84/esp32-micro-sdcard.
This library works well in a vanilla ESP32 and I can remap the pins. When I use the library with the T-call it doesn't work. I obtain always the message: Initializing SD card...initialization failed!
Does anyone have experience with an sd card module?

lewisxhe, 2019-09-06:
Refer to [here] (https://github.com/Xinyuan-LilyGO/TTGO-T4-T10-TestCode/blob/6c5e149eca3b084d3407538a85e8d7737033ef2f/TTGO_Vxx.ino#L165), all you need to do is specify the SPI pin @robottini

eueduardo3, 2019-09-09:
Which file can I reset the SPI pins on?

closed 2019-09-19 by vshymanskyy

jenil163, 2020-06-22:
> I'm trying to use an sd card module with the TTGO T-call. I can't use the SPI pins because I need to use also the SIM800L module. The SIM800L uses the standard pins 23 (MOSI) and 5 (CS). I tried to [...]

Same Problem I Try to change pin but initialization  failed

Conwon99, 2020-12-15:
> Using the help of friend @lewisxhe I was able to make it work, but it is very important that the pin CS is NOT pin 15, I lost a few days with this, because the CS as pin 15 the ESP32 is restarting forever
> 
> #include "FS.h"
> #include "SD.h"
> #include <SPI.h>
> SPIClass sdSPI(HSPI);
> //---------------------SD i used this pins
> #define SD_CS 5
> #define SD_SCLK 14
> #define SD_MISO 12
> #define SD_MOSI 13
> void setup() {
> Serial.println("Initializing SD card...");
> sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
> if (!SD.begin(SD_CS, sdSPI)) {
> 
> Serial.println("Card Mount Failed");
> 
> }else{
> Serial.println("DEU CERTO INICIAR SD CARD");
> }
> }

Although Isn't Pin 5 used as RESET for the SIM800L ?

lewisxhe, 2020-12-16:
RESET has been cancelled in subsequent versions, and it is not needed

ilker07, 2021-07-01:
I changed the pins but it gives me error

ilker07, 2021-07-01:
I tried this way.
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define SCK  18
#define MISO  19
#define MOSI  23
#define CS  5

SPIClass spi = SPIClass(VSPI);
  spi.begin(SCK, MISO, MOSI, CS);

  if (!SD.begin(CS,spi,80000000)) {
    Serial.println("Card Mount Failed");
    return;
  }
//I tried both 5v and 3.3 vcc  but it gives me error to.

carlosprato1, 2021-07-01:
i try with the same code   @eueduardo3  and probe with diferent pins and continue give a error.

and if i try Uncomment the line #define ENABLE_SPI_SDCARD from the code "Arduino_TinyGSM.ino" into in reboot loop.

i probed with two version of board "BLYNK 1.3v and 1.4v".

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13416
load:0x40080400,len:3672
entry 0x400805f8

[um⸮⸮⸮⸮2-hal-cpu.c:211]setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   840][D][esp32-hal-psram.c:84] psramInit(): PSRAM enabled
[  1180][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1180][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  1180][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  1185][E][sd_diskio.cpp:789] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1494][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  1494][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed

bilbo67000, 2022-06-12:
Hi. I am using a LILYGO T-CALL BLYNK V1.4 20-8-11
and, as many people, i am trying to work with a SDCard module.
Nothing I have read has worked :-)
I have tried different pins, trying to avoid pins already used by the TTGO 
but what I get is the SDCard module powered but the TTGO leds are OFF...
The setup function seams not be executed (nor a set ON Blue LED, nor a simple Serial.println message ).
The TTGO works fine without connecting the SdCard module ( Sending/receiving SMS).
This tells me something wrong on the connecting schema :-( 
Well... I can't try EVERY set of connecting pins, waiting for beeing lucky ;-) 

Exemples that doesn't work (assuming Ground and 3.3V ar connected to the SDCard module 
and the TTGO is connected to the computer USB-C)
pins 26, 27, 4, 32, 33, 23, 5 should not be used (used for the SIM800L), 13 (for the blue LED)

MISO   12   12   12
MOSI   18   18   25
SCK      15   14   14
SS        2      15  35

Here is an example  of the code tested (as I have disabled the call SIM800L functions  )
#include "FS.h"
#include "SD.h"
#include <SPI.h>
SPIClass sdSPI(HSPI);
//---------------------SD i used this pins
#define SD_CS xx
#define SD_SCLK xx
#define SD_MISO xx
#define SD_MOSI xx
void setup() {
Serial.println("Initializing SD card...");
sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
if (!SD.begin(SD_CS, sdSPI)) {

Serial.println("Card Mount Failed");

}else{
Serial.println("Card successfully mounted");
}
}

PLEASE HELP ... 
According to the ESP32 I use, with pins can I Use ?
Thanks