Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #21: micropython example still not working

fboerman, 2019-09-22:
even after the latest fix after issue #16 I dont get it to work. The gsm.atcmd('AT') gives back '\r\nOK\r\n' and the status is 89. It never leaves that. Im using the exact code that you supplied in the example. When I flash the board with tinygsm I can connect so my apn settings are correct.

kurogeek, 2019-09-25:
Hi, the result '\r\nOK\r\n' that was given back from gsm.atcmd('AT') means you are now able to connect to gsm module, not mobile network. I can't confirm that where the problem come from unless I see the whole gsm log.

fboerman, 2019-09-25:
well I tried it three times again with debug flag and suddenly it came alive. Not sure what happened before. thanks!

closed 2019-09-25 by fboerman

ciror00, 2020-02-20:
Hi! The example code does not work. The debug screen shows this:
```
Waiting for AT command response....I (1748) [PPPOS CLIENT]: AT COMMAND: [AT..]
I (1772) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (1872) [PPPOS CLIENT]: AT COMMAND: [AT+CFUN=4..]
I (1943) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (1943) [PPPOS CLIENT]: GSM initialization start
I (2543) [PPPOS CLIENT]: AT COMMAND: [AT..]
I (2565) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (2665) [PPPOS CLIENT]: AT COMMAND: [ATZ..]
I (2688) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (2788) [PPPOS CLIENT]: AT COMMAND: [ATE0..]
I (2811) [PPPOS CLIENT]: AT RESPONSE: [ATE0...OK..]
I (2911) [PPPOS CLIENT]: AT COMMAND: [AT+CFUN=1..]
I (2937) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (4037) [PPPOS CLIENT]: AT COMMAND: [AT+CPIN?..]
I (4062) [PPPOS CLIENT]: AT RESPONSE: [..+CPIN: READY....OK..]
I (4162) [PPPOS CLIENT]: AT COMMAND: [AT+CREG?..]
I (4186) [PPPOS CLIENT]: AT BAD RESPONSE: [..+CREG: 0,2....OK..]
W (4186) [PPPOS CLIENT]: Wrong response, restarting...
(very much logs)
.I (66811) [PPPOS CLIENT]: Skip command: [AT..]
I (66811) [PPPOS CLIENT]: Skip command: []     
I (66811) [PPPOS CLIENT]: Skip command: []     
I (66814) [PPPOS CLIENT]: Skip command: []     
I (66819) [PPPOS CLIENT]: Skip command: []     
I (66924) [PPPOS CLIENT]: AT COMMAND: [AT+CREG?..]
I (66948) [PPPOS CLIENT]: AT BAD RESPONSE: [..+CREG: 0,0....OK..]
W (66948) [PPPOS CLIENT]: Wrong response, restarting...
E (66949) [PPPOS CLIENT]: PPPoS TASK TERMINATED
......Traceback (most recent call last):
  File "main.py", line 50, in <module>  
Exception: Modem not responding!
```
Does anyone have any solution? Thanks!.