Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #24: Arduino_GSM_OTA - Write always fails

GeirFrimann, 2019-10-06:
I have been trying to do OTA update with the supplied sample Arduino_GSM_OTA several times, but it always fails at 95%:

`--------------------------
Build:    Oct  5 2019 20:14:24
Flash:    8192K
ESP sdk:  v3.2.3
Chip rev: 1
Free mem: 358748
--------------------------
IP5306 KeepOn FAIL
  Firmware A is running
--------------------------
78 - Starting OTA update in 10 seconds...
13082 - Initializing modem...
22463 - Modem: SIM800 R14.18
22463 - Waiting for network...
28591 - Connecting to GPRS
34704 - Connecting to mysite.se:443
37001 - Requesting /files/OTA/pellets_prod_poc/firmware.bin
40781 - Flashing...
 95%
No Error
106487 - FATAL: Write failed. Written 356100 / 357104 bytes`

Any suggestions?

s2shreyas, 2019-10-16:
Hi everyone , am also facing same issue, that after 95% its giving Fatal error.

GPRS Status:Connected
CCID:
IMEI:862631038855747
Signal quality:-110... -54 dBm 17
Battery level:100%
Battery voltage:4.98
76967 - Connecting to s3.ap-south-1.amazonaws.com:443
87642 - Requesting /fw.rtos.paqs/Blink1.ino.bin
89835 - Flashing...

5%
10%
15%
20%
25%
30%
35%
40%
45%
50%
55%
60%
65%
70%
75%
80%
85%
90%
95%
No Error
341570 - FATAL: Write failed. Written 170596 / 174992 bytes
ets Jun  8 2016 00:22:57

please suggest how to overcome from this issue

srivathsa-paqs, 2019-10-17:
> I have been trying to update my device with the supplied sample Arduino_GSM_OTA several times, but it always fails at 95%:
> 
> ```
> --------------------------
> Build:    Oct  5 2019 20:14:24
> Flash:    8192K
> ESP sdk:  v3.2.3
> Chip rev: 1
> Free mem: 358748
> --------------------------
> IP5306 KeepOn FAIL
>   Firmware A is running
> --------------------------
> 78 - Starting OTA update in 10 seconds...
> 13082 - Initializing modem...
> 22463 - Modem: SIM800 R14.18
> 22463 - Waiting for network...
> 28591 - Connecting to GPRS
> 34704 - Connecting to mysite.se:443
> 37001 - Requesting /files/OTA/pellets_prod_poc/firmware.bin
> 40781 - Flashing...
>  95%
> No Error
> 106487 - FATAL: Write failed. Written 356100 / 357104 bytes
> ```
> 
> Any suggestions?

GeirFrimann did u get any solution for this problem, evn i am getting same error

GeirFrimann, 2019-10-18:
@srivathsa-paqs No, no resolution yet :-(

joseluislucas, 2020-09-02:
Hi guys, I'm facing the same problem.

@GeirFrimann @s2shreyas @srivathsa-paqs did you get it working? Any suggestion on where to look to fix that?

Thanks

srivathsa-paqs, 2020-09-02:
> Hi guys, I'm facing the same problem.
> 
> @GeirFrimann @s2shreyas @srivathsa-paqs did you get it working? Any suggestion on where to look to fix that?
> 
> Thanks

Hi ,yes its solved , it is because of GSM signal quality (Bandwidth) , please replace antenna or take device where you have good internet speed. in our case we have replaced the GSM antenna.

grichfitz, 2020-10-08:
This is a signal issue. Change the antenna or move the unit.
 You should close this issue with this resolution.

Eiyvor, 2020-11-06:
This could also be due to insufficient keep-Alive timeout, try increasing the keep-Alive timeout in the server. sim800 use slow 2G network.

gunter72, 2021-01-18:
Hi, I have the same problem, I tried to act on the antenna but it doesn't change, do you have any other suggestions?

gunter72, 2021-01-18:
> This could also be due to insufficient keep-Alive timeout, try increasing the keep-Alive timeout in the server. sim800 use slow 2G network.

i tried with a light firmware, under 300k but i have the same problem.

closed 2021-01-19 by lewisxhe