Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #25: OTA_Fatal_Error

s2shreyas, 2019-10-16:
Hi everyone , am also facing same issue, that after 95% its giving Fatal error.

GPRS Status:Connected
CCID:
IMEI:862631038855747
Signal quality:-110... -54 dBm 17
Battery level:100%
Battery voltage:4.98
76967 - Connecting to s3.ap-south-1.amazonaws.com:443
87642 - Requesting /fw.rtos.paqs/Blink1.ino.bin
89835 - Flashing...

5%
10%
15%
20%
25%
30%
35%
40%
45%
50%
55%
60%
65%
70%
75%
80%
85%
90%
95%
No Error
341570 - FATAL: Write failed. Written 170596 / 174992 bytes
ets Jun  8 2016 00:22:57

please suggest how to overcome from this issue

grichfitz, 2020-10-03:
Same problem... Did you resolve?

grichfitz, 2020-10-08:
This is a signal issue. Change the antenna or move the unit.
 You should close this issue with this resolution.

closed 2021-01-19 by lewisxhe