Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #26: OTA fatal Error after 95%

shrujan-paqs, 2019-10-17:
I am using your library and trying to interface but its throwing error. Please help.

Here is the output:

GPRS Status:Connected
CCID:
IMEI:862631038855364
Operator:Airtel
Local IP:100.64.193.20
Signal quality:-110... -54 dBm 17
Battery level:100%
Battery voltage:4.98
76967 - Connecting to s3.ap-south-1.amazonaws.com:443
87642 - Requesting /fw.rtos.paqs/Blink1.ino.bin
89835 - Flashing...

5%
10%
15%
20%
25%
30%
35%
40%
45%
50%
55%
60%
65%
70%
75%
80%
85%
90%
95%
No Error
341570 - FATAL: Write failed. Written 170596 / 174992 bytes
ets oct  8 2019 00:22:57

grichfitz, 2020-10-03:
Same problem... Did you resolve??

grichfitz, 2020-10-08:
This is a signal issue. Change the antenna or move the unit.
 You should close this issue with this resolution.

closed 2021-01-19 by lewisxhe