Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #27: OTA Not completing TTGO-T-Call

srivathsa-paqs, 2019-10-17:
I am trying to download firmware from the given OTA example but after 95% it is Giving Error.

GPRS Status:Connected
CCID:
IMEI:857631038855652
Operator:Vodafone
Local IP:100.64.193.20
Signal quality:-110... -54 dBm 17
Battery level:100%
Battery voltage:4.98
76967 - Connecting to s3.ap-south-1.xxxxx.com:443
87642 - Requesting /fw.rtos.paqs/Blink1.ino.bin
89835 - Flashing...

5%
10%
15%
20%
25%
30%
35%
40%
45%
50%
55%
60%
65%
70%
75%
80%
85%
90%
95%
No Error
341570 - FATAL: Write failed. Written 170396 / 174991 bytes
ets Jun  9 2016 00:18:63

What could have caused the error, Please Give Solution as soon as possible.

grichfitz, 2020-10-03:
Same problem... Did you resolve?

grichfitz, 2020-10-17:
This is a signal issue. I resolved by getting a new aerial

closed 2021-01-19 by lewisxhe