Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #29: How to power the board?

jbrepogmailcom, 2019-10-22:
Hello,

I cannot find any documentation on how to power the TTGO-T-Call board. Do I connect 4V battery to 5V PIN, or to plastic connector on the board? Is the battery charged when connected to USB?

Thanks for answer, Jan

crash007, 2019-10-23:
There is a battery connector on the backside. Use the cable you got to
connect.

Den tis 22 okt. 2019 kl 21:45 skrev jbrepogmailcom <notifications@github.com

jbrepogmailcom, 2019-10-23:
OK, got it and it works. Two more questions please:
- is battery charged through that connector when TTGO is connected to USB-C?
- is there any way how to measure the battery power without resistor divider? I tried routine to measure internal voltage with  rom_phy_get_vdd33(), but that seems to measure power after stabilizer not battery

crash007, 2019-10-24:
Yes it charges battery when connected to usb-c
You can measure pin 35*2 to get voltage. I do like this
int batteryLevel = analogRead(35)*2;

Den ons 23 okt. 2019 kl 23:10 skrev jbrepogmailcom <notifications@github.com

jbrepogmailcom, 2019-10-24:
Works well, thank you. Multiplication by 2 should supposedly give me milivolts, but the value was too high. so I had to use something like analogRead(35)*2 / 1135 to get real voltage.

eueduardo3, 2019-10-24:
> Yes it charges battery when connected to usb-c You can measure pin 35*2 to get voltage. I do like this int batteryLevel = analogRead(35)*2; Den ons 23 okt. 2019 kl 23:10 skrev jbrepogmailcom <notifications@github.com
> […](#)

I can input 9V in usb-c ?

emigbur, 2019-10-24:
Hi
about battery measurement I took some samples for more accuracy measurement, I compared with a tester and is very accurate.
void getBateryValue() {
  // tomar un número de muestras analógicas y sumarlas
  while (sample_count < NUM_SAMPLES) {
    sum += analogRead(BATTERY_METTER_PIN);
    ample_count++;
    delay(10);
  } 
   voltage = (sum/NUM_SAMPLES)/ 2350.0;
  batteryLevel = _min(map(sum / NUM_SAMPLES, 2000, 2440.0, 0, 100), 100); //1100
  if (sum/NUM_SAMPLES < 1200) batteryLevel = 0; 
  Serial.print("Valor Analogo: ");
  Serial.println(sum);
  Serial.print("Battery Lavel:");
  Serial.print(batteryLevel);
  Serial.println("%");
  sample_count = 0; //Reset counter
  sum = 0;  //Reset counter
  voltage = voltage * 4.20;
  Serial.print(voltage);
  Serial.println (" V");
 }

jbrepogmailcom, 2019-11-26:
my experiences:

I had great variations with batteryLevel = analogRead(35), so I am averaging 20 measures 0.1s apart. That seems to be pretty stable.

I had issues with powering the board with LiPo 3.6V battery via the connector. It works rarely and it seems that the board is not getting enough power. Probably the transition resistance of about 1 ohm in battery holder and thin wires is too much for high current. Anyway, I have connected battery directly to pins GND and 5V and the board seems to work well. The voltage returned by batteryLevel = analogRead(35) is now about half of that when battery connected via battery connector, so I will have to adjust it.

I would like to ask for opinion - can the charging circuitry cause that the board is getting less power from the battery?

AF340, 2020-01-19:
Hello, 
Is there a way to optimize the battery consumption? with a 3.7v 2000mAh li-ion Lipo, it discharge in 24hrs, without doing anything special (web server running waiting for client and send every 15mns a http req)

sarusso, 2020-02-14:
If you need to have a web server always on I don't think you can do much. If  you drop the web server and only send an http request every 15 minutes, you can then use the deep sleep mode (which is more a controlled power off than a sleep) and you should be able to save consistent amounts of energy. See here: https://github.com/Xinyuan-LilyGO/TTGO-T-Call/issues/40 and here: https://github.com/Xinyuan-LilyGO/TTGO-T-Call/pull/46 (where the contributor measured ~1mA in deep sleep mode)

Der2k7, 2020-03-02:
Why not simple power with 3x 1.5V AA batteries on 5V pin? In deep sleep it consumes than 100-150uA.

jbrepogmailcom, 2020-03-03:
Did you actually measure the current? For me it is several mA even in deep sleep mode

Der2k7, 2020-03-03:
Yes, I measure the current. Board consume in deep sleep about 120uA (3x AA batteries on pin 5V).

jbrepogmailcom, 2020-03-03:
Please post the .ino script or link to it that you use to put TTGO to sleep. Thank you very much

Der2k7, 2020-03-03:
```
#define MODEM_POWER_ON 23
...
...
...
void GoToSleep() {
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  digitalWrite(MODEM_POWER_ON, LOW);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_start();
}
```

Der2k7, 2020-03-03:
@nikil511
I see you have a lot of experience with this.
What do you think about powering the board (pin 5V) with three AA batteries (if we don't use rechargeable batteries), do you have any concerns about proper functionality (GSM spikes,...)? Is the 5V pin suitable for this? 
Do you found maybe any other optimal solution to operate with the rechargeable Li-Ion batteries and minimum power consumption in deep sleep mode (up to 200uA)?
Our need is simple, GSM modem send data once an hour, in the meantime is in deep sleep. And as long as possible without charging the batteries or any other intervention ;-)

jbrepogmailcom, 2020-03-06:
Dear Der2k7, thanks for example.
It did not work, because it was using obsolete esp_deep_sleep_pd_config, which is now esp_sleep_pd_config. After I fixed the script, I got to 1250uA when I blinked LED, without LED code I got 950uA. Therefore I think it is not possible to achieve less than 1mA and definitely not 120uA even when powering directly to 5V rail by 3 AAA cells as you suggest.

Here is modified code that works:
`
#define MODEM_POWER_ON 23

#define TIME_TO_SLEEP 3600
#define uS_TO_S_FACTOR 1000000ULL

#define LED_BUILTIN 13

void setup() {
  // put your setup code here, to run once:

esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
pinMode(MODEM_POWER_ON, OUTPUT);
digitalWrite(MODEM_POWER_ON, LOW);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);

pinMode(LED_BUILTIN, OUTPUT);

}

void loop() {
  // put your main code here, to run repeatedly:

digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
delay(1000);                       // wait for a second
digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
delay(1000);                       // wait for a second

esp_deep_sleep_start();

}
`

grichfitz, 2020-04-21:
Hay,
Can someone confirm that you can power this board with 5v onto the 5v pin?
Everything I seem to find says you cannot do this, but it would make my project a lot easier!
TIA

maromme, 2020-04-21:
i do  works fine. just take care you've got 2A on your supplies

BillyGriffiths, 2020-04-30:
> i do works fine. just take care you've got 2A on your supplies

 @maromme With 5V supplied to the pins directly, does it charge the LIPO battery?

maromme, 2020-04-30:
didn't try that yet but regarding comments of other users here, it's not charging in that case

BillyGriffiths, 2020-04-30:
> didn't try that yet but regarding comments of other users here, it's not charging in that case

Would have been great if it could. I'm running the board powered using AC/DC PSU 220V->5V@3A and it would have been amazing if the battery could then serve as backup power on a mains power failure. I would need to figure out how to get the USB-C powered via my existing power source.

nikil511, 2020-04-30:
with a USB-C cable maybe? :-)

kyklist, 2020-05-09:
What lithium  batteries are good to use (or to avoid) for TTGO Lilygo T-Display??

I put a NCR18650 via the small connector on the back and it works.
When the battery is somewhat discharged part of the board near the connector gets hot while charging. 

Cheers,
Richard L.

vitalstates, 2020-06-13:
I've been struggling for documantation on this.
Can anybody tell me how to power the baord from the usb socket and supply seperate power(not battery) to the the sim800 part with 2a capability. I intend to run from 12v supply and use lm2596 for voltage drop. In this scenario there would be no lipo and no lipo charging involved.

thanks

Ed

nikil511, 2020-06-13:
If you provide 5v via USB , that should be enough for stable gsm operation.
Depending on PSU maybe you might have to add a capacitor too on the 5v line

You can also power them directly via 5v pin, but I had a few tcall powered
long term that way and some failed.

Either way, new tcall has an i2c power managment chip so behavior will be
different.

vitalstates, 2020-06-13:
Many thanks for quick response....
I was thinking that the usb supply from the PC would not be enough current for the gsm during upload and debugging so thought a seperate 4v 2a would be necessary for the sim800.

nikil511, 2020-06-13:
(usb is enough)
But..why don't you try it first, before making your life complicated?

maromme, 2020-06-13:
hi nikil,
could you please be a bit more detailed there:
"directly via 5v pin, but I had a few tcall powered long term that way and some failed."

nikil511, 2020-06-13:
There is (was) a major bug with IP5306 that upon power failure and
recovery, it needed "reset" to start.  I am solar charging them thus to
overcome this problem I tried powering tcall with a solar-charger module
that had 5v output (e.g. dfrobot, seeedstudio, adafruit,sparkfun etc) It
worked for a few weeks, but then my devices went offline. Many things could
have gone wrong, so I do not have a diagnosis and I dont think its
important anymore, since tcall power management has completely changed-
improved in the latest version.

grichfitz, 2020-06-14:
Hay nikil511,
You may beable to shine some light on an issue Im having. 
I'm having a problem with my T-Call, powered directly via 5vpin.
On GPRS connection, I send two HTTP post successfully and then disconnect. The T-Call restart on its own.
Sending data via wifi, same method, I have no problems, only when sending via SIM it reboots.
Any advice... help will be appreciated, its driving me up the wall.
G

nikil511, 2020-06-14:
If esp32 does a brown out reset during gprs transfer or registration then
you are underpowered, a capacitor might help.
If the restart is after successful http request, then check your code! It
does not sound like a power problem.

thetandberg, 2020-06-16:
I have used both "batteryLevel = analogRead(35)*2;" and "battVoltage = modem.getBattVoltage();" from the TinyGsmClient library, to get the battery voltage. The strange thing is I get a reading of about 4 volts, wether I have a battery connected or not. Can anyone explain?
I also have a few other questions. 
Is there a way in software to find out if the battery is being charged? 
Can the battery supply power to the esp32 as well, or only to the SIM800 modem?

nikil511, 2020-06-16:
4v is the power to the gsm module via regulators.

See the examples to understand how to get the battery

thetandberg, 2020-06-18:
Thank you for quick response @nikil511 :)

What examples are you refering to? The "int batteryLevel = analogRead(35)*2;" is an example in this thread.

Shouldn't the reading be zero, when no battery is connected?

Do you also have the answer to the other two questions?

Eckaard, 2020-06-28:
Sorry if this question has already be asked. I'm currently working with the Lilygo t-call programmed using micropython. My issue is as follows. When I power the device using a 18650 battery connected to the back attachment, I am unable to switch the device on. I switch it off by holding down the reset button for 10 seconds however can't switch it on again. Is there a gpio pin that can be used to do this or am I just doing something wrong. Currently to switch on the system I insert the USB-c cable.

eugvb, 2020-07-29:
I've issues when power T-call(IP5306) by 5v pin with 220->5v 3A supply.
Esp32 reseting with Guru Meditation Error (IllegalInstruction). 
Are there any solutions? What max 5v pin voltage? May be i can increase input to 5.5v and this solve problem?

nikil511, 2020-08-02:
Add capacitor, and/or change psu
You shouldn't have to go above 5v

eugvb, 2020-08-07:
I use PSU Meanwell RS-15-5 with output 5V 3A, if I connect this PSU to USB, it works very well.
I doesn't try add capacitor, what capacity it should be?

P.S. In newer T-Call with AXP192 power pin marked as "5V-IN", does it mean, that connecting to this pin is simmilar to usb and I'll can charge battery?

anku2424, 2020-08-10:
Hi, I connected ttgo tcall with 2 Li ion batteries in series though the attachment on its back. So in total I provided it near about 7.4 V supply. But it didn't work and now it's not even working with usb. Does ttgo tcall not accept this voltage? Is there any way to rectify this? Thanks in advance.

nerukoma, 2020-12-01:
Hello there, regarding the board power. 

If I want to connect 4x18650 v4 batteries, can I connect them through the plastic pin located in the other side of USB-C connector. That would be around 7.4V 5000mAh. 

Is there any discharge/overcharge protection inside the board?

Will a USB-C charge the batteries when it would be connected? (for example 24 hours per 3 months).

Thank you

bhupiister, 2021-01-13:
> Hello there, regarding the board power.
>
> If I want to connect 4x18650 v4 batteries, can I connect them through the plastic pin located in the other side of USB-C connector. That would be around 7.4V [...]

Did you try this ?

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

Ronel308, 2021-02-11:
Good day,I need a code to switch two relays attached to the T-Call SIM800 please.
I have search the internet with no results!

bhupiister, 2021-02-11:
@Ronel308 Your issue is not related to the product. Simply search how to connect relay to esp32. And if you want to learn how to send command over gsm internet or SMS to switch relay on and off, simply search randomnerdtutorials for that. This all is very basic stuff. Moreover, esp32 pins can output 0V (low) and 3.3V (high) and you can search for relay boards with switching voltage of 3.3V, for example this one. [https://robu.in/product/2-channel-isolated-5v-10a-relay-module-opto-coupler-arduino-pic-avr-dsp-arm/](url)

Ronel308, 2021-02-15:
Good day,
I am new to this enviroment and 
would like to control a relay from 
this board from my cellphone.
I have access to my own server.
Kind regards,
Kobus

Kobus Steenkamp
Cell 0828937098

Barabas19, 2021-03-22:
> Hello there, regarding the board power.
>
> If I want to connect 4x18650 v4 batteries, can I connect them through the plastic pin located in the other side of USB-C connector. That would be around 7.4V [...]

You can connect just 1S battery to the battery connector. If you want to use 4 batteries, just connect them in parallel.
Discharge/overcharge protection is implemented.
The battery can be charged only via USB connector. 5V pin does not charge the battery!

weyersma, 2021-03-28:
Hi. I am able to read the battery voltage on PIN35. I want to read if the USB is powered or not. Is there a way to read the power on the USB to detect if the Power is lost?

tewbc, 2021-04-04:
> Hi. I am able to read the battery voltage on PIN35. I want to read if the USB is powered or not. Is there a way to read the power on the USB to detect if the Power is lost?

I used to see the voltage on PIN35 for my case. If USB power is missing, the voltage will drop below 4V.

Barabas19, 2021-04-05:
> Hi. I am able to read the battery voltage on PIN35. I want to read if the USB is powered or not. Is there a way to read the power on the USB to detect if the Power is lost?

That's possible to obtain from IP5306 power manager. It is connected to ESP32 via I2C.
Datasheet: http://www.injoinic.com/wwwroot/uploads/files/20200221/0405f23c247a34d3990ae100c8b20a27.pdf
Registers: https://docviewer.yandex.ru/view/0/?page=8&*=H5f3ndGZgnTZYCmrKmDgl66huU57InVybCI6InlhLWRpc2stcHVibGljOi8vZTlnMElGZVlCM3ZwaTlxbklETy9wTThHZzRXOVhCc1F4UmRqNHZqYzNxVEJtay9COU1EVkFpV1BUTS9wVFR6QnEvSjZicG1SeU9Kb25UM1ZvWG5EYWc9PSIsInRpdGxlIjoiSVA1MzA2IEkyQyByZWdpc3RlcnMucGRmIiwibm9pZnJhbWUiOmZhbHNlLCJ1aWQiOiIwIiwidHMiOjE2MTc2Mzg2ODM1NTUsInl1IjoiNTkwNDQ2MTM4MTU3OTgwMTc0MCJ9

Register 0x70 bit 3 is that you are looking for.

weyersma, 2021-04-06:
> That's possible to obtain from IP5306 power manager. It is connected to ESP32 via I2C.
> Datasheet: http://www.injoinic.com/wwwroot/uploads/files/20200221/0405f23c247a34d3990ae100c8b20a27.pdf [...]

thank you for this. it makes sense. I am struggling with my implementation but logically that makes a lot of sense. will keep chipping away at it

rynskyi, 2021-05-21:
Thanks to the information collected here
I wrote a [small class for controlling the IP5306 through I2C](https://github.com/rynskyi/IP5306)
Maybe someone will find this useful 
(pull request welcome)

ind2016ltd, 2021-11-06:
Hi,

Just started using TTGO T-CALL SIM800L v1.4 and it got '5v out' pin instead of 5v, is safe to power it up via that pin? Have used previously v1.3 and had no issues powering it up via 5v pin. Just got confused by that 'out'. 
No battery charge is required. 
Any experience is welcome.

rytisp, 2021-11-06:
Hi,

Just started using TTGO T-CALL SIM800L v1.4 and it got '5v out' pin instead of 5v, is safe to power it up via that pin? Have used previously v1.3 and had no issues powering it up via 5v pin. Just got confused by that 'out'.
No battery charge is required.
Any experience is welcome.

LilyGO, 2021-11-06:
[reference](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800#2-about-the-5v-power-pin-description-on-the-board)， 5vOUT is output 5V. Can't power t-call.

rytisp, 2021-11-06:
Is any way to order bigger quantities of v1.3 somewhere, seems like it's discontinued and everyone just selling v1.4.

LilyGO, 2021-11-08:
Which version 1.3 do you refer to? Can you send me a picture?There are boards with AXP192 that can use 5V input.[Refer to purchase link](https://pt.aliexpress.com/item/4001274909689.html)

Barabas19, 2021-11-08:
I power it using [USB-C plug](https://www.aliexpress.com/item/1005001457790699.html?spm=a2g0s.9042311.0.0.27424c4dFjHYo7)

rytisp, 2021-11-08:
> I power it using [USB-C plug](https://www.aliexpress.com/item/1005001457790699.html?spm=a2g0s.9042311.0.0.27424c4dFjHYo7)

Me too. But just for temporary use. I need to power it via the pin, usbc for me is not solution.

rytisp, 2021-11-08:
> Which version 1.3 do you refer to? Can you send me a picture?There are boards with AXP192 that can use 5V input.[Refer to purchase link](https://pt.aliexpress.com/item/4001274909689.html)

This v1.3
[image]

LilyGO, 2021-11-09:
Output 5V when usB-C is connected.Do not connect USB - C.Version 1.4 can also power the board through 5V.

[24 more replies]