Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #30: ESP IDF PPPoS Causing Fatal Crash

trishaulr, 2019-11-07:
I've updated the RX and TX pins in the config, and have added code to set up pins 5, 4 and 23. However, a fatal crash still occurs when running the program.

Error Dump:
E (6949) esp-modem: esp_modem_dte_send_cmd(215): process command timeout
E (6949) dce_service: esp_modem_dce_sync(48): send command failed
E (6949) sim800: sim800_init(460): sync failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d43bb  PS      : 0x00060530  A0      : 0x800d2639  A1      : 0x3ffb62e0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffb51f0  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x800d43b9  A9      : 0x3ffb62b0  
A10     : 0x00000000  A11     : 0x3ffbae4c  A12     : 0x400d55f8  A13     : 0x3ffb6308  
A14     : 0x00000000  A15     : 0x0001c200  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000007c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

slechta, 2020-02-08:
The problem is missing check for return value of 
`modem_dce_t *dce = sim800_init(dte);`
call.

The crash is a result of NULL return value, which is then used in the next line:
`ESP_ERROR_CHECK(dce->set_flow_ctrl(dce, MODEM_FLOW_CONTROL_NONE));`

Fix the crash like this:
```
    modem_dce_t *dce = sim800_init(dte);
    if (dce == NULL) {
        ESP_LOGE(TAG, "SIM 800 initialization failed");
        return;
    }
```

This fixes the crash, but the modem will not work. The problem is the you send the AT command and get no response. This results in timeout which will result in sim800_init returning NULL.

One of the reasons for that is, that you have wrong pin configuration:
Go to file named **sdkconfig** and set
```
CONFIG_EXAMPLE_UART_MODEM_TX_PIN=27
CONFIG_EXAMPLE_UART_MODEM_RX_PIN=26
CONFIG_EXAMPLE_UART_MODEM_RTS_PIN=23
CONFIG_EXAMPLE_UART_MODEM_CTS_PIN=25
```
I have also made a nice overview of the PINs on the board here:
https://docs.google.com/spreadsheets/d/1En2lLH3iiQ3bMoiHqdnGr9lFwZXz_NWt59ze-k06Mek/

I hope it helps. I am also trying to get the GSM work, I am facing exactly the same issue as you.

slechta, 2020-02-08:
The other catch is, even if the model is not using any UART FLOWcontrol, you still need to specify the RTS and CTS pin. These pins will be reconfigured even if there is no flow control.

in esp_modem.c
```
    if (config->flow_control == MODEM_FLOW_CONTROL_HW) {
        res = uart_set_pin(esp_dte->uart_port, CONFIG_EXAMPLE_UART_MODEM_TX_PIN, CONFIG_EXAMPLE_UART_MODEM_RX_PIN,
                           -1, -1);
    } else {
        res = uart_set_pin(esp_dte->uart_port, CONFIG_EXAMPLE_UART_MODEM_TX_PIN, CONFIG_EXAMPLE_UART_MODEM_RX_PIN,
                           UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    }
```

Now look into the uart_set_pin method:
```
    if(rts_io_num >= 0) {
        PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[rts_io_num], PIN_FUNC_GPIO);
        gpio_set_direction(rts_io_num, GPIO_MODE_OUTPUT);
        gpio_matrix_out(rts_io_num, uart_periph_signal[uart_num].rts_sig, 0, 0);
    }
    if(cts_io_num >= 0) {
        PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[cts_io_num], PIN_FUNC_GPIO);
        gpio_set_pull_mode(cts_io_num, GPIO_PULLUP_ONLY);
        gpio_set_direction(cts_io_num, GPIO_MODE_INPUT);
        gpio_matrix_in(cts_io_num, uart_periph_signal[uart_num].cts_sig, 0);
    }
```
This code will touch the RTS and CTS pin configuration even if there is no flow control.

Hackaroth, 2020-04-09:
Hi!
I have the same problem. After fix the crash the problem is still there. 

E (972) esp-modem: esp_modem_dte_send_cmd(215): process command timeout
E (972) dce_service: esp_modem_dce_sync(48): send command failed
E (972) sim800: sim800_init(460): sync failed
E (972) pppos_example: SIM 800 initialization failed

Someone have solve it?

Thanks

slechta, 2020-04-12:
Well, I have. I have a working MQTT connection. Code is here
https://github.com/slechta/esp-idf/tree/gsmfix

eflukx, 2020-05-10:
Yes, similar problem here. I have working code here, but when modem cannot be properly synced with (e.g. when not properly reset but still in [ppp/data] session) the fatal crash occurs (when a graceful fail is expected). I checked for NULL-checks in code close to the application code, alle seems normal. Possibly be something in the deeper Modem/PPP layers (?)

Ps. In my case I use a BG96 device, so the problem does not seem to lie in modem-specific parts.

k4mils, 2020-12-04:
> The problem is missing check for return value of
> `modem_dce_t *dce = sim800_init(dte);`
> call.
>
> The crash is a result of NULL return value, which is then used in the next line: [...]

Hello @slechta . I looked at your Google document with T-call pins review and noticed a connection between IO2 pin and GSM Modem reset. Where did you get this information from? I haven't seen such a connection on schematic.

closed 2021-01-19 by lewisxhe