Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #31: Cannot flash

Mesmer88, 2019-11-09:
When trying to flash with platformIO i'm getting this error:

Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

On some other boards i would have kept BOOT button pressed but T-call doesn't have one.
Any hints?

Mesmer88, 2019-11-11:
Update, seems to be working but very low success rate: like 2 in 10 attempts if i keep resetting board during flashing.

lepetitcreole, 2019-11-27:
Hi, I have the same issue! Flashing rarely works on my board too, I need to try more 10 times before to make it work.

slechta, 2020-01-08:
Same issue here. The first day everything was fine and second day I am getting:
```
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

cfficaurzua, 2020-01-22:
After a lot of errors, i found out that the trick is to press reset 3 times between the last dot and the dashes. (with rhythm ♩♩♩)
e.g: 
**Connecting**........_____..... (click here 3 times) ____
Now the code always uploads.
why? ¯\_(ツ)_/¯
how to fix it? ¯\_(ツ)_/¯

slechta, 2020-02-08:
When I disconnected the module from the daughterboard (self-made) the flashing started working. So either there was some kind of short-circuit or there was parasite capacitance on some pin.

slechta, 2020-02-08:
Note: Threre is no need to push any button. The UART CTS/RTS signals are able to reset the board to download mode even if there is already sotware running.

roberto1984, 2020-02-22:
Hi,

i have the same issue, i selected the ESP32 dev module board but it does not flash, i receive a timeout..which are the parameters to set on arduino after the ESP32 dev board has been selected?(i mean for instance flash mode,flash size, flash frequency and so on..)
Thank you in advance

maromme, 2020-02-22:
pin 0 to ground puts the board in download mode. that's necessary at least if you program not using USB. bug well, give it a try

roberto1984, 2020-02-23:
Thanks maromme!it did the trick!!!!

maromme, 2020-02-23:
you're welcome.
for complete description:  it's necessary to to remove the connection of pin 0 to ground andpress the reset button after  download

JB-DX, 2020-04-17:
I had the same problem. The reason is the very weird structure of RTS/DTR circuit.  IO0 jumps high exactly when EN goes high. So ESP32 never sees IO0=0 at start.
Delaying EN a bit solves that. A 1uF from EN to ground fixes it.

vicival, 2020-05-20:
> **Delaying EN a bit solves that**.    A **1uF from EN to ground** fixes it.
  
  Thanks!

jfchiantera, 2020-06-13:
if you connect io0 to GND and press reset you receive in serial port 

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

closed 2020-07-01 by lewisxhe

mariusgrams, 2020-07-17:
Hi guys, I have the problem too, but I can't get it fixed. I connected GPIO 0 to GND, but nothing has changed. 

[image]

I always get the error: Failed to connect to ESP32: Timed out waiting for packet header.
Could someone explain in more detail how they fixed the problem?

jacoherotel, 2021-08-28:
> After a lot of errors, i found out that the trick is to press reset 3 times between the last dot and the dashes. (with rhythm ♩♩♩)
> e.g:
> **Connecting**........_____..... (click here 3 times) ____
> Now the code always uploads.
> why? ¯_(ツ)_/¯ how to fix it? ¯_(ツ)_/¯

works like a charm thnx

reddy9698, 2021-12-08:
Hi guys I have almost the same problem. The module was working perfectly and suddenly the upload started failing at 0x00034864 address. I have tried all the methods and there's no difference.
`Compressed 230000 bytes to 121170...
Writing at 0x00010000... (12 %)
Writing at 0x0001f010... (25 %)
Writing at 0x00024f87... (37 %)
Writing at 0x0002c228... (50 %)
Writing at 0x00034864... (62 %)

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2`