Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #33: SIM card no inserted problem

Ecancan, 2019-11-15:
```
#include <SoftwareSerial.h>
#define RXD2 16
#define TXD2 17
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22
//SoftwareSerial mySerial(MODEM_RX, MODEM_TX); // RX, TX
#define SerialAT Serial1

void setup()
{
  Serial.begin(115200);
  //Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
 //mySerial.begin(9600);
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);

  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);a
  digitalWrite(MODEM_POWER_ON, HIGH);
}

void loop()
{
  if (SerialAT.available())
    Serial.write(SerialAT.read());
  if (Serial.available())
    SerialAT.write(Serial.read());
}
```

Result : 

> AT
> 
> OK
> AT+CPIN?
> 
> ERROR
> AT+CMEE=1
> 
> OK
> AT+CMEE=2
> 
> OK
> AT+CPIN?
> 
> +CME ERROR: SIM not inserted
>

lewisxhe, 2020-07-01:
Please use the Arduino_TinyGSM sample to verify, if the problem has been solved, please close the problem

toph84, 2020-12-05:
Hi,

same issue here (+CME ERROR: SIM not inserted) with a new unit of SIM800C AXP192 VERSION 20200609. I tried a second SIM-card (which work well in 2G cell phone and the SIM800L IP5306 VERSION 20200811) as well as different sketches (Arduino_TinyGSM example and simple AT-command-test-sketch).

Is it a hardware problem?

Best wishes
Chris

closed 2021-01-19 by lewisxhe