Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #34: SIM800 Audio

Krezalis, 2019-11-27:
Anyone have reference of connection (SP)eaker and (MIC)rophone directly to board? I want make some intercom: speaker and microphone up to 100 meters. 
And more one question: can sim800 encode/decode audio and stream it to serial port?

AF340, 2019-12-19:
Same question, MIC and SP that work with my A6board don't with TTGO CALL

Duncro, 2020-01-18:
I also have issues with the audio. The microphone is unusably soft and the speaker just doesnt work. I used the same mic and speaker I sue on other sim 800 modules. I noticed the board has the filtering circuits surface mounted, thought they might be faulty but even after removing them the mic is still terrible and the speaker completely inoperable. Before removing them i tried boosting their gains with AT commands, didn't work. Starting to think these pins aren't even connected

trdsr, 2020-05-22:
It is really frustrating to see the pins SPK+ and SPK- on the board and then they don't work :-(

mykobzar, 2020-06-16:
It is works. Just send 

```
AT+CLVL=100
AT+CMIC=1,15
AT+CAAS=0
AT+CHF=0,1
```
for set it up.

Duncro, 2020-06-18:
> 
> 
> It is works. Just send
> 
> ```
> AT+CLVL=100
> AT+CMIC=1,15
> AT+CAAS=0
> AT+CHF=0,1
> ```
> 
> for set it up.

Thanks @mykobzar! That works!

trdsr, 2020-09-13:
The speaker works now with these settings, but I can't get the microphone to work.
@Duncro , have you been successful with the mic?

mykobzar, 2020-09-13:
Both, mic and speaker work.
I'm using microphone from headphones. But any electret mic will work.
Check polarity. They have "+" and "-".

Duncro, 2020-09-13:
Yup, both worked with those settings. Have you tested the mic on something else?

mykobzar, 2020-09-13:
I got mic from broken but the working headphones. But any other have to work. Just connect a contact which is connected to mic case to "-", another one to "+".

trdsr, 2020-09-14:
Thank you for your answers. I tried 3 microphones (and switched polarity) but no success. Maybe I have a faulty module.
This is the micropython code I use:

```python
from machine import Pin, UART

MODEM_PWKEY_PIN = 4
MODEM_RST_PIN = 5
MODEM_POWER_ON_PIN = 23
MODEM_TX_PIN = 26
MODEM_RX_PIN = 27

def initPins():
    MODEM_PWKEY_PIN_OBJ = Pin(MODEM_PWKEY_PIN, Pin.OUT)
    MODEM_RST_PIN_OBJ = Pin(MODEM_RST_PIN, Pin.OUT)
    MODEM_POWER_ON_PIN_OBJ = Pin(MODEM_POWER_ON_PIN, Pin.OUT)
    MODEM_PWKEY_PIN_OBJ.value(0)
    MODEM_RST_PIN_OBJ.value(1)
    MODEM_POWER_ON_PIN_OBJ.value(1)

def gsmSetup():
    sendATCommand('ATE0')
    sendATCommand('AT+CSCLK=0')
    sendATCommand('AT+CLVL=100')
    sendATCommand('AT+CMIC=1,15')
    sendATCommand('AT+CAAS=0')
    sendATCommand('AT+CHF=0,1')

def sendATCommand(cmd, value=None):
    if value is None:
        uart.write(cmd + "\n")
        print("sending", cmd)
    else:
        uart.write(cmd + value + "\n")
        print("sending", cmd + value)
    out = uart.read()
    return out if out is None else out.decode('utf-8')

uart = UART(1, 9600, timeout=1000, rx=MODEM_TX_PIN, tx=MODEM_RX_PIN)
gsmSetup()
```

lewisxhe, 2020-09-18:
The problem seems to be resolved and will now be closed!

closed 2020-09-18 by lewisxhe