Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #36: Power Using 5V pin

ShashikaTECH, 2019-12-25:
I'm trying to power the board and charge the battery using 5v pin. (without using type-c input) is this possible, do i have to turn off the 5v boost to use this.

UCIS, 2020-04-18:
I have tried this as well. Unfortunately the 5V pin is connected to the output of the battery management IC, so you can not use it to charge the battery. You can draw 5V from this pin for other components while the board is running (on battery or USB-C power).

I could not find a way (on the board or in the schematic) to charge the battery, other than via the USB-C port. We'll likely use a USB C connector with only VCC and GND connected.

grichfitz, 2020-04-22:
Hya,
So are you confirming you can power the board via the 5v pin without issues?
I’m trying to get confirmation regarding this as there is little documentation on this. 

The schematics I’ve seen suggested power from the 5v goes to the USB, but there is a diode in between. Which suggests this won’t work. 
TIA

maromme, 2020-04-22:
As I wrote in #29 it work's fine to power the bord. I've got 6 boards running fine with 5V/2A at the 5v pin. And I don't use any battery at the battery plug.

But be careful powering at the 5v pin AND using the USB port. there are reverse powering issues that might damage the power source at the 5v pin. just happened to me.

grichfitz, 2020-04-22:
Thanks maromme, sorry I missed your other post, just read it. 
Do you know if this goes through the onboard regulator, I assume it does, just asking. 
Soz to hear about your board... did you kill it!?

maromme, 2020-04-22:
sorry but there i can't help you. didn't really bother about that. 
I just killed a cheap regulator. not one of the nice pololu. I'm using the D24V22F5 even though these are a bit expensive the idle power consumption is very low and that counts for me

ShashikaTECH, 2020-04-22:
Thank you all. Its been a long time since i asked the question. Now I have 2 boards and both of them are configured to stay in deep sleep mode with 5v boost on to power some PIR sensor to wake the board and send a message. (I use type C to charge the battery) This worked for about a month continues but after that  the 5v boost fail to wake up and power everything. I tried t

UCIS, 2020-04-22:
> So are you confirming you can power the board via the 5v pin without issues?

Yes, that seems to work, but as said before keep in mind that when powering from USB C or battery, the 5V pin will output power, which might damage another source connected directly to the 5V pin.

> The schematics I’ve seen suggested power from the 5v goes to the USB, but there is a diode in between. Which suggests this won’t work.
It's actually a 2A fuse according to the schematic. The labeling is not very clear, there are 3 different nets: VBUS (the USB input), 5V (the fused VBUS input - used to power the IP5306 and USB-serial chip) and +5V (which is the boost 5V output connected to the external 5V pin).

So the external 5V pin is actually the net labeled +5V, connected to the output of the IP5306 and can **not** be used to charge the battery, but it can power the rest of the board.

aryvini, 2020-05-12:
Is it better to power the board using some 18650 Li-ion batteries in parallel to the battery connector, or to build a circuit with 5V regulator and power directly to the 5V pin?
Btw, what means the MIC+/- and SPK+/- pin on the board? I tried to find some info on the schematics but could not find it.

UCIS, 2020-05-14:
@aryvini it depends on whether you want to use the charger function. If you want to use the charger function, you have to connect the batteries to the battery connector. If you do not need this, it could be better to connect a regulated source to the 5V pin with additional buffering capacitors. Others wrote that the battery connector may be a little weak to provide the required peak power.

The MIC and SPK pins connect to the SIM800L microphone and speaker functions, this allows you to connect a microphone and speaker for a regular audio call. You'd have to dial a phone number or accept an incoming call using AT commands.

grichfitz, 2020-06-20:
Is there a recommended GND pin if you are using 5v pin?

I’m finding the T-Call is more stable using the GND closet to the 5v pin. Has anyone else experienced this?

lewisxhe, 2020-07-01:
I have verified that I can use 5V to input 5V directly and it works well

closed 2020-07-01 by lewisxhe