Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #37: TTGO-T-Call - What is the best Power supply with a Battery?

schmeckm, 2020-01-05:
Dear ALL,

I want to use this board for an IoT Use Case.   Do you have suggestion how to supply the Board best with power? I can operate the board over the USB Board to connect GPRS Network. 
The use case consider a GPS Sensor, OLED 0.96 Display and Temperature sensor. Each of them need 5 Volt Power. 
In manual is written that it's possible to use the battery plug on the Board to supply the board with power or to use the 5 Volt Pin, but I have my concerns that it would be enough to supply the sensor as well. I suppose that the board will consume like other GSM board some peak currents (1-2 A) to start up GSM and I suppose that LIPO which is direct connected to the Board will not supply the Sensor enough with power. 
Do you have any suggestion or circuit schema example how the power supply should be built up? 
Thanks your your advice in advance,

Markus

grichfitz, 2020-04-21:
Hay Markus,
Hope you are well.
Did you have an answer to your question regarding powering the board?
Im looking to power the TTGO via the 5v pin, but Im no 100% sure this is possible and would like some clarifcation before I try!
Hope to hear from you,
Richard

scubastation, 2020-05-19:
A LiPo with 3.7 V on the 5V works perfect!

closed 2020-07-01 by lewisxhe