Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #39: How to detect charging complete?

jbrepogmailcom, 2020-01-22:
Hello, since there is no charging LED, is it possible to detect when did T-Call stop charging? For example by reading some IC5306 register by I2C? It would be good if T-Call can send a message when its battery has been charged and device can be unplugged.

ClemensGruber, 2020-01-26:
I found only awful english documentation of the IC5306 Is there any register information in english, the one here in github is paritally englisch, most Chinese!

sarusso, 2020-02-17:
@jbrepogmailcom, @ClemensGruber and anyone  ending up here: the chip referred to as "IC5306" is actually an "IP5306" (typo?); to detect if battery is charged either read the pin 35 via ADC (i.e. see here: https://github.com/Xinyuan-LilyGO/TTGO-T-Call/issues/29#issuecomment-558632092) or read the IP5306 via I2C (i.e. see here: https://github.com/Xinyuan-LilyGO/TTGO-T-Call/pull/46/commits/8e4eb04b47988b2e8ee1a6b328cf1321f9b01487#diff-3cee23a9527dd91490e16b092f062144)

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

bhupiister, 2021-01-21:
[image]

For anyone who needs to know how are power pins and battery pins connected. Also, I was not aware of the onboard divider circuit to read voltage. I hope this helps someone.