Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #40: High deep sleep mode current?

jbrepogmailcom, 2020-01-22:
When T-Call is in sleep mode (RTC GPIO on in order to receive interrupt)powered by battery connector and SIM800L powered down, it draws about 5.5mA. It seems a lot to me. I checked voltage chips in the schematics. They are drawing about 100uA each, then there is a number of voltage dividers, that draw 20uA each. ESP32 in deep sleep shoud be about 10-150uA. When I sum it up, total draw should be still less than 1mA. But I have over 5, what could be wrong?

ClemensGruber, 2020-01-26:
I measured a power consumption of 850 uA in deep sleep with 

    esp_sleep_enable_timer_wakeup(1000000 * TIME_TO_SLEEP_IN_S);
    esp_deep_sleep_start();

But this is still 3x the ammount you can read on the [product page](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1127)
> Sleep current:  About 300uA

I added also the deep sleep code above to the TinyGSM example: 

https://github.com/Xinyuan-LilyGO/TTGO-T-Call/blob/master/examples/Arduino_TinyGSM/Arduino_TinyGSM.ino

but also no success, still 850 uA, I disabled `bool isOk = setPowerBoostKeepOn(1); ` in the hope that this reduces power consumption and tried an additional `digitalWrite(MODEM_POWER_ON, LOW);` assumint that pin 23 switches off the buck converter SY8089 that powers the the SIM800 but no success with both trials! 

Any idea how to reduce power maximal in deep sleep??

jbrepogmailcom, 2020-01-27:
I think there is no simple way to turn off power converters. You even can not turn off IP5306 programatically, because its KEY pin (where you would send LOW for more than 2s) is also connected to reset of ESP32. You can program IP5306 to switch off 32s after ESP32 goes to sleep, but then you can only wake it manually by pressing reset. I was trying to find a way how to shut down everything and just keep ESP32 in sleep powered directly to 3.3V rail, but a) power was still about 1mA, b) I could not wake the board only manually

maromme, 2020-02-06:
I'm powering my t-call with +5v from a buck converter straight to the 5v rail.  works fine. by chance I found out that i can reduce the sleep power consumption by around 50%: 
- send the t-call to deep sleep
- pull the +5v connector for a short moment
- after that the deep sleep consumption is reduced
- if you pull the GND connector board restarts with high consumption. 

I'm not done with testing yet.  possible reason for that behavior could be rtc ds3231 "back powering" the board. if I find time the next days, i can post a circuit of the setting

jbrepogmailcom, 2020-02-06:
Does not "pull the +5v connector for a short moment" turns off ESP32 completely?

maromme, 2020-02-06:
well, that's what one would expect...
but as the ds3231 got a battery there is a second power source in the circuit. small but maybe enough? I'm curious myself just found that out the other day. just need some time for testing.
nice thing: the reduced consumption stays low even in the next deep sleep cycle.

maromme, 2020-02-07:
I got some time:

It's a thing with the RTC DS3231. Feeding the T-call with 5V at the 5V pin:

1) The plain T-Call in DeepSleep (wake up source: button at GIPO32 ) takes about 0,1mA
2) adding the RTC DS3231 it's 13,4mA
3) if I pull the +5V wire from the T-call for a moment I'm down to 2,8mA -> the t-call is not waking up or restarting (!)
4) waking up by button push (it's the same with wake up by RTC), go to sleep again -> still 2.8mA
5) removing the power source for a moment (rtc AND T-call wirthout power) -> 13.4mA 

conclusions:
A) the t-call is getting enogh power during deep sleep via the rtc that one could pull the power wire of the t-call
B) the short power interruption at the t-call brings the power consumtion of the rtc down

I don't know why, but well that's probalbly not the right thread for that question (even if I'm really curious)

jbrepogmailcom, 2020-02-08:
DS3231 seems to take a lot - do you power it through pins? I think it may have optimized power.in through battery contacts, it should take something like 0.01mA. Together with 0.1mA of ESP32 it would be pretty good. I also think it would be better to power ESP32 to 3.3V rail rather than to 5V, but I had pretty high current in that case, over 5mA in deep sleep.

ClemensGruber, 2020-02-11:
> It's a thing with the RTC DS3231. Feeding the T-call with 5V at the 5V pin:
> 
>     1. The plain T-Call in DeepSleep (wake up source: button at GIPO32 ) takes about 0,1mA

Did you measure 0.1 mA, (means 100 uA!) or was this a typo? I got only a bit less than 1 mA in deep sleep!

maromme, 2020-02-11:
for the Ds3231 I gonna follow these suggestions https://thecavepearlproject.org/2014/05/21/using-a-cheap-3-ds3231-rtc-at24c32-eeprom-from-ebay/

in particular i'll remove two resistors and leave the VCC pin floating. means powering the rtc from the coin battery only

I'm quite sure that I measured 0.1mA but can validate that again. it's just the t-call without anything. measured in the +5V line directly at the t-call

ClemensGruber, 2020-02-12:
> I measured a power consumption of 850 uA in deep sleep with

In addition to that: I measured this power consumption direct after going to deep sleep. I had an T-Call running since some days and power consumtion seems to be "too low" for the postulated 850 uA so I measured in an extended way, means a time longer than directly after going sleeping only. And I observed a step down after around 40 seconds the microcontroller went to sleep. After that time I measured 500 uA deep sleep current. So a part, the SIM800 or the IP5306 ... needs some time to turn off any consumer. Still not the mentioned 300 uA on the website but a good ammount underneath the first mesured 850 uA!

jbrepogmailcom, 2020-02-12:
Yes but can you wake the ESP32 after that automatically? I do not think so, because actually after 32 seconds the IP5306 turns off and you cannot wake it automatically, only by pressing reset. I would say that all other circuitry can take those 850uA when ESP32 is actually _turned off_.

ClemensGruber, 2020-02-12:
Yes, I can wake up the T-Call after that, but I do not cut power totally with an external timer. LiPo is always connected to the board. I tried different sleep times, 2 hours are possible without any problems. 

But I'm using the setPowerBoostKeepOn(int en) function like in the example code.

lewisxhe, 2020-02-15:
IP5306 has a light load automatic shutdown function, so if you need to keep the esp32 sleep state and use the esp32 internal timer to wake up, you must turn off the IP5306 light load power off function. As for writing 300uA on the page, I think that should be wrong. It seems that the power consumption of one version of the board has not been well optimized. A large part of the reason is due to power issues. I will collect opinions and try to solve them in the next version.

sarusso, 2020-02-18:
I am not sure if this is the right place where to ask, but I don't understand if it is possible and how to disable the IP5306 light load automatic shutdown function (and having the T-Call board going in deep sleep on battery power avoiding the famous 32 secs no return timeline). This is not related to the setPowerBoostKeepOn function mentioned several times in this thread, right? I also find mixed messages on other boards using the IP5306 (i.e. https://github.com/m5stack/M5Stack/issues/156) and I still don't understand if light load automatic shutdown can be turned off or not, and if yes how. Does anyone have a clearer view? Thanks!

sarusso, 2020-02-18:
I am not sure if this is the right place where to ask, but I don't understand if it is possible to disable the IP5306 light load automatic shutdown function (and thus having the T-Call board to wake up from deep sleep on battery power after the famous 32 secs no return timespan).

This is not related to the setPowerBoostKeepOn function mentioned several times in this thread, right? I Googled a lot and found mixed messages also with respect to other boards using the IP5306 (i.e. https://github.com/m5stack/M5Stack/issues/156) and I still don't understand if light load automatic shutdown can be turned off or not, and if yes how.

Does anyone have a clearer view? Thanks!

closed 2020-09-18 by lewisxhe