Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #43: HTTPS post  fail

prasadshi, 2020-01-27:
Https post call returning 400 response code using TTGO T1 sim800l TinyGsm library

KelvinPac, 2020-02-18:
Have you verified the information you are sending on the post request or the server is actually okay.?

prasadshi, 2020-02-19:
Thank you , It works fine now with different network sim

Regards
Prasad

biswajitjei02, 2020-06-14:
> Thank you , It works fine now with different network sim Regards Prasad
> […](#)

HI
 can you help me with this ?
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800L/issues/71

KelvinPac, 2020-06-15:
Hi. Are you still having the 400 response code issue? Please Note `400 Bad Request response status code indicates that the server cannot or will not process the request due to something that is perceived to be a client error (e.g., malformed request syntax, invalid request message framing, or deceptive request routing)` If still getting a 400. Please try calling the endpoint using a REStful client like Postman or Talend and verify you get the response you expect. I suspect you are writing the request the wrong way which is very easy to do when using the HTTP calls in arduino

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

soniritesh124, 2022-04-26:
same error plz help