Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #44: GPS Connection

schmeckm, 2020-02-02:
Hi All
If I want to connect to this Board a GPS Sensor like NEO-8M, what kind of pins shall I use?

Markus

anthonyvage, 2020-02-03:
Connect your GPS Module to Pin 32 and 33 (you can use other pins too)
`HardwareSerial SerialGPS(2);
....
SerialGPS.begin(9600, SERIAL_8N1, **32**, **33**);
`

slechta, 2020-02-08:
There is no GPS. It is GSM/GPRS only.

slechta, 2020-02-08:
Definitely use the hardware UART. It is more reliable as there is UART queue of events and interrupts available out of box.

DroneUniversity, 2020-03-01:
Dear All,

I am also having a problem. I have connected a TF Mini Lidar to the default RX and TX pins of the board and I do get silly alphanumeric results on my Serial Monitor.

The problem for me that I am having trouble in understanding is that I already have a "Serial2.begin(115200);" command in void setup, which is used for connecting the dev board to Cayenne dashboard.

How could I setup alternative pins for reception of the Lidar's output and not get confused with the already Serial2.begin(115200) command I have?

Thank you in advance
Spyros

UCIS, 2020-04-18:
As this board uses the ESP32 chip, which has a GPIO matrix feature, you can use almost any pin (combination) for any peripheral. You can use Serial0, 1 and 2 simultaneously with any of the available GPIO pins using `Serial0/1/2.begin(9600, SERIAL_8N1, rx_pin, tx_pin);`. Just make sure you do not use an input only pin for TX.

You can even change the pins assigned to Serial0/1/2 later in your application by just calling `begin` again, and use the same HardwareSerial peripheral to communicate with different external components.

llaneiro, 2020-09-02:
Guys, I'm having a problem trying to connect a NEO-6M
When I try to declare ANY port as RX and TX from the GPS, sim800l is no longer able to connect to the internet, he just keeps trying to connect but never succeeds. While I declare the same ports as sim800l (26 and 27), it works again, but I don't think I can use them for GPS since they are those of sim800l. 

What should I do in this case? How did you succeed in declaring the serial for a GPS?

rajeshatuce, 2020-09-06:
I am also facing similar problem, I am not getting any output of Neo6m in my serial monitor. Please let me know if any issue with code:
// Please select the corresponding model

 #define SIM800L_IP5306_VERSION_20190610

// Define the serial console for debug prints, if needed
#define DUMP_AT_COMMANDS
#define TINY_GSM_DEBUG          SerialMon

#include "utilities.h"

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to the module)
#define SerialAT  Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800          // Modem is SIM800
#define TINY_GSM_RX_BUFFER      1024   // Set RX buffer to 1Kb

#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

HardwareSerial SerialGPS(2);

#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  15        /* Time ESP32 will go to sleep (in seconds) */
#define SERVER_CONNECTION_TIMEOUT 10000
#define APN_CONNECTION_TIMEOUT 10000
#define HTTP_CONNECTION_TIMEOUT 10000L

// Server details
const char server[] = "abc.co.in";
const char resource[] = "/test";

// Your GPRS credentials (leave empty, if missing)
const char apn[]      = "airtelgprs.com"; // Your APN
const char gprsUser[] = ""; // User
const char gprsPass[] = ""; // Password
const char simPIN[]   = ""; // SIM card PIN code, if any

TinyGsmClient client(modem);
const int  port = 80;

#include "FS.h"
#include "SD.h"
#include <SPI.h>

SPIClass SPI1(HSPI);

#define MY_CS       33
#define MY_SCLK     25
#define MY_MISO     27
#define MY_MOSI     26

void setupModem()
{
#ifdef MODEM_RST
    // Keep reset high
    pinMode(MODEM_RST, OUTPUT);
    digitalWrite(MODEM_RST, HIGH);
#endif

[... 169 lines cut ...]
    delay(10);

    // Start power management
    if (setupPMU() == false) {
        Serial.println("Setting power error");
    }

    // Some start operations
    setupModem();

    // Set GSM module baud rate and UART pins
    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
    SerialGPS.begin(9600, SERIAL_8N1, 32, 33);//Neo6m Rx goes to pin 32 and Tx goes to Pin 33. Tried reversing but no luck
}

void loop()
{
    if (SerialGPS.available()) {
   SerialMon.print(char(SerialGPS.read()));  // read from gps, write to serial debug port
  } else{
    SerialMon.println("Not available");
  }
    //initiateGPRSSubmitHTTPRequest();
    //initiateDeepSleep();
}

closed 2021-01-19 by lewisxhe