Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #47: Does the board support switching from USB power to battery without reset?

mpflanzer, 2020-02-10:
I have a fully charged battery connected to the board and the boost output of the IP5306 is enabled (SYS_CTL0=0x37). Still, whenever I disconnect the USB-C cable the board resets. After the reset it is running fine on battery power though. Is that the normal behavior? I expected that I could just unplug the USB cable and it would seamlessly transition to being powered from battery. Maybe I'm missing a setting for the IP5306?

lewisxhe, 2020-02-15:
Currently, IP5306 cannot seamlessly switch from USB to battery power. This issue is expected to be resolved in the next version.

delta010, 2020-02-29:
Can I use any battery with 3.7v . because when I connected a 12v battery with a voltage regulator to the ttgo t - call it don't even power on

maromme, 2020-02-29:
I use a 12v battery with 2A regulator down to 5v at the 5v pin of the t-call. works fine

think the battery connector can just be used for the recommended lithium batteries

maromme, 2020-02-29:
that's a nice one: 
low quiescence consumption. brings my setting down to about 1 mA at 12v in deep sleep

maromme, 2020-02-29:
that's a nice one: 
https://www.pololu.com/category/192/d24v22fx-step-down-voltage-regulatorslow quiescence consumption. brings my setting down to about 1 mA at 12v in deep sleep

abesis, 2021-01-02:
> I have a fully charged battery connected to the board and the boost output of the IP5306 is enabled (SYS_CTL0=0x37). Still, whenever I disconnect the USB-C cable the board resets. After the reset it [...]

I have managed a seamless transition from USB to battery, by adding a 4700μF capacitor between GND and 3.3V of the board. Probably a smaller capacitor will also be ok, but I havent tried.

bhupiister, 2021-01-13:
> I have managed a seamless transition from USB to battery, by adding a 4700μF capacitor between GND and 3.3V of the board. Probably a smaller capacitor will also be ok, but I havent tried.

I am planning on doing the same. I need to power device from usb and once charged, need to remove the usb. Just like mobile phones.  Is it a full proof solution? Or some other solution you found as well.

closed 2021-01-19 by lewisxhe