Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #48: pin header distance

t-xigit, 2020-02-13:
[Here ](https://de.aliexpress.com/item/33045221960.html) is a drawing of the dimensions of the board.

However, I am designing a PCB with this board and there is no information of the distance between these two pin header rows.
[image]

Krezalis, 2020-02-13:
1.27mm

t-xigit, 2020-02-13:
I don't mean the pitch of the header.
I mean the distance between the left and right header row.
Cheers

maromme, 2020-02-13:
I got that if it helps. in width its about 10.5 x 2.54mm = 26.67mm

[image]

If you are drwaing a footprint, I wold be grateful if you share it. Could need it myselfe.

t-xigit, 2020-02-13:
@maromme here you go mate
[ttgo-t-call-v1_3.zip](https://github.com/Xinyuan-LilyGO/TTGO-T-Call/files/4198896/ttgo-t-call-v1_3.zip)

maromme, 2020-02-14:
@t-xigit thank you very much. looks really good. looking forward to get my PCBs

t-xigit, 2020-02-14:
@maromme 
Be careful with the orientation of the board.
There is a description in the footprint section.
Other than that this design was tested. Just the pins were off by 1.5 mm

lewisxhe, 2020-06-28:
If there are still problems, please reopen the issue

closed 2020-06-28 by lewisxhe