Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #49: Available SPI Pins

myiotcenter, 2020-02-14:
Refer to readme.md, these pins are used for SIM800L :
```
SIM800_TX 26
SIM800_RX 27
SIM800_PWKEY 4
SIM800_RST 5
SIM800_POWER 23
````
Refer to ``esp32-wrover_datasheet_en.pdf`` following are the SPI Pins :
```
VSPI
VSPID / MOSI : IO23 --> SIM800L Power
VSPIQ / MISO :  IO19
VSPICLK /SCK  :  IO18
VSPICS0 /SS   :  IO5 --> SIM800L RST

HSPI
HSPID / MOSI : IO13 --> Built-in LED
HSPIQ / MISO : IO12
HSPICLK / SCK  : IO14
HSPICS0 / SS   : IO15

FSPI
SPID / MOSI : SD1 
SPIQ / MISO : SD0
SPICLK / SCK  : CLK
SPICS0 / SS   : CMD

Notice:
1. GPIO12 is internally pulled high in the module and is not recommended for use as a touch pin.
2. Pins SCK/CLK, SDO/SD0, SDI/SD1, SHD/SD2, SWP/SD3 and SCS/CMD, namely, GPIO6 to GPIO11 are connected to the SPI flash integrated on the module and are not recommended for other uses.
```
Having the fact that SIM800L uses 2 VSPI pins, HSPI MOSI also used by built-in LED, and FSPI are not recommended to be used, can someone please advise how can I have additional SPI connection to TTGO TCall Board?

Thanks !

t-xigit, 2020-02-14:
If you are using Micropython there is a Software SPI.
[https://docs.micropython.org/](https://docs.micropython.org/en/latest/esp32/quickref.html#software-spi-bus)

lewisxhe, 2020-07-01:
SD example has been added, SPI works well

closed 2020-07-01 by lewisxhe

tavdog, 2022-04-30:
where is the SD example ?  Do I have to reasign MOSI pin for SPI since it is used for SIM 800 PWR ?