Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #50: Micropython example still does not work yet

ciror00, 2020-02-20:
Hi! I come from the issue number #21 

The example code does not work. The debug screen shows this:
```
Waiting for AT command response....I (1748) [PPPOS CLIENT]: AT COMMAND: [AT..]
I (1772) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (1872) [PPPOS CLIENT]: AT COMMAND: [AT+CFUN=4..]
I (1943) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (1943) [PPPOS CLIENT]: GSM initialization start
I (2543) [PPPOS CLIENT]: AT COMMAND: [AT..]
I (2565) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (2665) [PPPOS CLIENT]: AT COMMAND: [ATZ..]
I (2688) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (2788) [PPPOS CLIENT]: AT COMMAND: [ATE0..]
I (2811) [PPPOS CLIENT]: AT RESPONSE: [ATE0...OK..]
I (2911) [PPPOS CLIENT]: AT COMMAND: [AT+CFUN=1..]
I (2937) [PPPOS CLIENT]: AT RESPONSE: [..OK..]
I (4037) [PPPOS CLIENT]: AT COMMAND: [AT+CPIN?..]
I (4062) [PPPOS CLIENT]: AT RESPONSE: [..+CPIN: READY....OK..]
I (4162) [PPPOS CLIENT]: AT COMMAND: [AT+CREG?..]
I (4186) [PPPOS CLIENT]: AT BAD RESPONSE: [..+CREG: 0,2....OK..]
W (4186) [PPPOS CLIENT]: Wrong response, restarting...
```
`(very much logs)`
```
.I (66811) [PPPOS CLIENT]: Skip command: [AT..]
I (66811) [PPPOS CLIENT]: Skip command: []     
I (66811) [PPPOS CLIENT]: Skip command: []     
I (66814) [PPPOS CLIENT]: Skip command: []     
I (66819) [PPPOS CLIENT]: Skip command: []     
I (66924) [PPPOS CLIENT]: AT COMMAND: [AT+CREG?..]
I (66948) [PPPOS CLIENT]: AT BAD RESPONSE: [..+CREG: 0,0....OK..]
W (66948) [PPPOS CLIENT]: Wrong response, restarting...
E (66949) [PPPOS CLIENT]: PPPoS TASK TERMINATED
......Traceback (most recent call last):
  File "main.py", line 50, in <module>  
Exception: Modem not responding!
```
Does anyone have any solution? Thanks!.

lewisxhe, 2020-06-28:
Please check the power supply, whether it is greater than 2A, and your power cord, please use a high-quality USB cable

ciror00, 2020-11-23:
Hi @lewisxhe !

Thank you very much for your answer.

The problem is intermittent. Sometimes it works and sometimes it doesn't. The cable is high quality and is the same as it works (sometimes).

shaunmulligan, 2020-12-04:
I have this problem consistently. If I switch the device to use the Arduino tinyGSM example with the same cable and setup it seems to work correctly so it seems like maybe something specific to the mPy example. Any ideas on how to trouble shoot this?

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

shaunmulligan, 2021-01-21:
Sorry @lewisxhe can we re-open this issue, as you can see from my message above I have having problems. The GSM module works fine with arduino tinyGSM code but fails with the micropython example which means that it doesn't seem to be a power issue but rather an issue with the micropython code. Can you give me ideas on how to troubleshoot this or another python library/code to use instead?

lewisxhe, 2021-01-21:
@shaunmulligan Have you run the MicroPython code in the example?  If possible, please publish your code and let me see.

lewisxhe, 2021-01-21:
@shaunmulligan Which version are you using?  Please let me know, I will test it again tomorrow.  By the way, please share the error log with me.

lewisxhe, 2021-01-22:
I tested the four boards currently on sale. Only one of them had an accident. The reason may be that the USB current was too small. At the moment of entering the network, the current was too large, causing the module to be triggered by an unexpected low battery. The solution is on the board. It can be solved by connecting a 1000mA lithium battery to the lithium battery interface. Hope this can help you. I have tested the program and there is no problem.

closed 2021-01-28 by lewisxhe