Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #51: Invoking AT+HTTPSSL command fails

amotl, 2020-02-21:
Hi there,

similar but different to #43, we are observing the `AT+HTTPSSL` command would fail completely. We've also reported this to https://github.com/pythings/Drivers/issues/3 and discussing it within [1] ff.

Invoking `AT+CGMR` yields firmware revision SIM800L16_20141031_1027 for the SIM800 on our T-Call device.

Thanks already for looking into this and with kind regards,
Andreas. 

[1] https://community.hiveeyes.org/t/unlocking-and-improving-the-pythings-sim800-gprs-module-for-micropython/2978/8

cc @ClemensGruber, @poesel

RafOXxxx, 2020-03-20:
Hi @amotl ,

could you solve your https-errors? I am experiencing similar issues. 
I am also using the pythings driver, the modem has the firmware revision **1418B05SIM800L24**. 
Weirdly:

`AT+HTTPSSL?`
yields
`+HTTPSSL: 0`
`OK`

`AT+HTTPSSL=1`
yields
`AT+HTTPSSL=1`
`ERROR`

but

`AT+CIPSSL=1`
yields
`AT+CIPSSL=1`
`OK`

I thought that if `AT+CIPSSL=1` works, `AT+HTTPSSL=1` should work as well?

ClemensGruber, 2020-03-21:
No we gave up (but don't stop reading here)! Seems that HTTPS support ist not available via old SIM800 modules, we had both szenarios a "older" T-Call bought autum last year, that does not support HTTPS with SIM800 R13.08 and a newer one with SIM800 R14.18 that supports HTTPS, both TTGO T-Calls!! See https://community.hiveeyes.org/t/unlocking-and-improving-the-pythings-sim800-gprs-module-for-micropython/2978/19

But switching to PPP seems to be an solution, see https://community.hiveeyes.org/t/unlocking-and-improving-the-pythings-sim800-gprs-module-for-micropython/2978/20

> But this is one more reason to switch over to PPP when running on an ESP32 and no longer depend on this old IP stack on the SIM800.

RafOXxxx, 2020-03-21:
Thanks for that information, I thought I may did something wrong or that may device is broken.

fabioce, 2020-07-11:
I have TTGO T-Calls with SIM800 R14.18 and https don't works

closed 2021-01-19 by lewisxhe