Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #55: AT+HTTPSSL=1 fails while AT+CIPSSL=1 yields OK

RafOXxxx, 2020-03-20:
Hi,

my SIM800L has following firmware revision **1418B05SIM800L24**.
I have been having trouble to do https-requests, I only got response status-code 603 with no content.
If I try to enable https like this:

`AT+HTTPSSL=1`

the modem response with 

`AT+HTTPSSL=1`
`ERROR`

while 

`AT+CIPSSL=1`

yields

`AT+CIPSSL=1`
`OK`

that confuses me a little bit, because a lot of sources say if  AT+CIPSSL is okay AT+HTTPSSL should also work. Could there be a problem with the certificates on the modem?

fabioce, 2020-07-11:
For me it's the same:
AT+HTTPSSL=1
ERROR
AT+CIPSSL=1
OK
I have TTGO T-Call with sim800L

aGGreSSiv, 2020-08-19:
how do you send AT commands? Im newbie. I would like to change some parameters. I used Sofware Serial. When I type AT on Arduino IDE console, nothing happens.. What should i do ? Produck link is [](https://www.banggood.com/LILYGO-TTGO-T-Call-V1_3-ESP32-Wireless-Module-GPRS-Antenna-SIM-Card-SIM800L-Board-p-1527048.html?rmmds=myorder&cur_warehouse=CN)

peteakalad, 2020-08-19:
Tinygsm example:
modem.sendAT("+CNETLIGHT=0");

Don’t need to include AT

aGGreSSiv, 2020-08-19:
Maybe I should ask more specifically. I have a phone that I don't use, instead of registering imei (I don't know if this device is compatible for my project.) I want to change the imei address with the following command. 

> AT + SIMEI = "1111111111111" 

I used this code to  solve my problem. But I couldnt. I do not know if I am sharing my problem / Question in the right place. Sorry if I'm sharing in the wrong place.

`#include <SoftwareSerial.h>

SoftwareSerial swSer(27, 26, false, 256);

void setup() {
  Serial.begin(115200);
  swSer.begin(115200);

  Serial.println("\nSoftware serial test started");

  for (char ch = ' '; ch <= 'z'; ch++) {
    swSer.write(ch);
  }
  swSer.println("");

}

void loop() {
  while (swSer.available() > 0) {
    Serial.write(swSer.read());
  }
  while (Serial.available() > 0) {
    swSer.write(Serial.read());
  }

}`

closed 2021-01-19 by lewisxhe