Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #56: Board could not reset on power

dzungpv, 2020-03-25:
I have 4 boards, but 1 could not auto reset after power, must manual press the button or it will not work both the EPS and SIM800, i check the EN circuit but it is normal.
How to repair the unit?

lewisxhe, 2020-06-22:
Have you solved this problem? If it has not been resolved, please let me know, otherwise please close this question

dzungpv, 2020-06-22:
> Have you solved this problem? If it has not been resolved, please let me know, otherwise please close this question

It not solve, i have 4 but only this one. Try replace the resistor but not work

lewisxhe, 2020-06-22:
Can't reset automatically on power-up. Generally, the automatic reset circuit is broken. Please check these two components.
[image]

closed 2020-09-18 by lewisxhe