Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #58: How to get PWM-pin activated on incoming call?

JB-DX, 2020-04-20:
I need a H/W signal that gets activated during an incoming call, like RI. But RI isn't wired to the ESP. Instead I hoped the PWM pin could be used as it is accessible via the "motor" pad.
But this is not activated even when AT+CBUZZERRING=1 is set.
Playing with AT+SPWM= command doesn't generate a PWN signal at all. It only gets set to Low or High at specific values. like these:
AT+SPWM=0,1000,1
AT+SPWM=0,0,0

Is the PWM and CBUZZERRING functionality missing in the SIM800L firmware?

lewisxhe, 2020-06-22:
I tried it and it does not start BUZZER, so it should not be supported! In the next version (IP5306), I have requested to connect RI, DTR, STATUS, these three pins to ESP32.

lewisxhe, 2020-06-28:
If there are still problems, please reopen the issue

closed 2020-06-28 by lewisxhe