Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #59: Kind of SIM card needed

JMC1973, 2020-05-05:
I’ve a SIM card with a data plan and sms plan but without voice call plan. (Thingsmobile)
It’s impossible to get connection.
Is this kind of sim supported?
Thanks

lewisxhe, 2020-06-17:
Support separate data, as long as it supports the frequency band

Mr-HaleYa, 2020-06-24:
@JMC1973 
literally any sim card will work if it supports 2g and 3g. I use Twilio but hologram.io is nice as well

Please close if your issue is solved

closed 2020-07-01 by lewisxhe

JMC1973, 2020-07-01:
Thank you, I have the board working. It’s mandatory use user and password to configure the apn.

Mr-HaleYa, 2020-07-01:
not if your APN doesn't require it. As I said I use Twilio and all I have to do is pop it in and set the APN to them.
glad you resolved your issue