Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #63: There is an error in the schematics (pdf) of the IP5306 power management SoC

vicival, 2020-05-19:
IP5306's pin 5 (BAT) goes in reality to the R56 - C56 joint.
See the attached picture.
[image]

lewisxhe, 2020-06-17:
Feedback has been given, but the notification is correct
[image]

closed 2020-07-01 by lewisxhe

nekuneko, 2021-05-05:
I don't understand why there are two different shematics for the same IP5306 IC and the R4 resistor's value is different.
[image]

nghia09081999, 2024-01-12:
How do I make Vout always output 5V when I want to turn the KEY pin on and off many times?
[image]

osmany-ferrer, 2024-09-01:
Hello there,

My issue has to do with IP5306 battery sensing. LED 4 (100% full) remains blinking forever after battery is charged, as if it does not correctly detect the battery voltage. Wiring is according to the schematic (V1.31) and also the battery is ok because it gets fully charged in another charger.
In version V1.31 of the datasheet,  pin 6 (BAT) is wired to the battery through the resistor R4. However, in version V1.01, pin 6 (BAT) is directly wired to the battery.  I wonder if I should be guided by the previous version?

Thanks in advance for your help,
 Osmany

nghia09081999, 2024-09-03:
After a while the light will turn off if your circuit is directly connected
to the Battery Cell.

osmany-ferrer, 2024-09-03:
Thanls for your answer. I'll give it a try.
Best bregards,
Osmany