Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #65: Library for upgraded power management AXP192?

BillyGriffiths, 2020-05-20:
According to the documentation, power management on my SIM800L v1.3 board is handled by AXP192. Is there a library available for getting data like getUSBVoltage() etc?

peteakalad, 2020-05-31:
Seen this example sketch?
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800L/blob/master/examples/Arduino_TinyGSM_AXP192/Arduino_TinyGSM_AXP192.ino

Uses this library for reading AX192 vbus voltage, vbus current and batt voltage;
https://github.com/lewisxhe/AXP202X_Library

BillyGriffiths, 2020-06-01:
> Seen this example sketch?
> https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800L/blob/master/examples/Arduino_TinyGSM_AXP192/Arduino_TinyGSM_AXP192.ino
> 
> Uses this library for reading AX192 vbus voltage, vbus current and batt voltage;
> https://github.com/lewisxhe/AXP202X_Library

Exactly what I was looking for! Not sure how I missed it. Thanks @peteakalad !

closed 2020-06-01 by BillyGriffiths