Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #66: Ground issues with GPRS and deepsleep

grichfitz, 2020-05-20:
Hya everyone,
I´m experiencing some strange issues with my project and I wondered if anyone else has experience of this.

My project goes into deepsleep and randomly gets wake up with ext1 from any of the pins I have assigned to ext1. Sometimes it gets into a fit and continues waking up.

The project is stable when there is no SIM and using Wifi.

Each of the wake up pins have 10k pulldown resistors.

I´m wondering if it a ground issue??

Any thoughts??

TIA

lewisxhe, 2020-06-17:
What you need to check is your code, whether the used pins are used

grichfitz, 2020-06-17:
Are you using a breadboard? I had similar issues of it waking up randomly, spent a week trying to diagnose. 
In the end having a 10k resistor directly connected from the same breadboard connection as the wake up pins to ground solved the problem. 
I use 12,13,14,15 and 25 for my wake up. What pins are you using?

grichfitz, 2020-06-17:
This was solved in the end by making sure the 10k resistor was connected to the same connection in the breadboard as the GPIO pin, direct to gnd.

closed 2020-06-17 by grichfitz