Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #69: Where can you buy the updated version

Gary-cat, 2020-06-04:
Hi

I am very happy that there is an upgrade to the product. Where can I purchase it, and be sure I get the new version. I have 6 of the old version and happy with them, but I need more. 

Thanks Gary

kickouille, 2020-06-09:
Same here, I ordered one from BG but still got the older version, and I am unable to find the new one which would fix my issue #70...

Gary-cat, 2020-06-09:
Contact them at sales@lillygo.cc

I actually managed to get them on chat via the website, and bought 10 of them. That should be here soon. Specifically asked for the new ones. And they understood and processed correctly. They are a bit more expensive than the older ones though, but not by much.

Ali express could not ship to my country for some dumb reason. So they DHL it.

Gary-cat, 2020-06-09:
Sorry email is sales@lilygo.cc

peteakalad, 2020-06-09:
I ordered recently from lilygo store on AliExpress here;
https://www.aliexpress.com/item/33045221960.html

Yet to arrive but I hope its the newer AXP192 based board to avoid the need to press the button every time its powered?  If not they really ought to put two products on AliExpress so you can choose.

lewisxhe, 2020-06-17:
Now LilyGo sells the latest version with power management. The old version has been eliminated. As for whether it will start automatically when powered on, it is best to communicate with the customer service before inquiring.

lewisxhe, 2020-06-17:
Now the version sold by LilyGo can be selected on the purchase page. The old version has been eliminated. As for whether it will start automatically when powered on, it is best to communicate with the customer service before inquiring.

closed 2020-06-28 by lewisxhe