Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #71: How to use HTTPS rest api in LilyGo-T-Call-SIM800L?

biswajitjei02, 2020-06-14:
here is the example of HTTPS GET /POST as this data base shows please check
https://firebase.google.com/docs/reference/rest/database.

how to use https request using Tiny GSM library for data base .
tiny gsm library https://github.com/vshymanskyy/TinyGSM

lewisxhe, 2020-06-17:
TinyGSM provides a sample program with https
https://github.com/vshymanskyy/TinyGSM/blob/master/examples/HttpsClient/HttpsClient.ino

fabioce, 2020-07-11:
With TTGO T-Call sim800L https don't work

lepetitcreole, 2020-08-29:
HTTPS doesn't work for me too. HTTP call is ok but HTTPS calls fail...

mihai-lazar, 2020-08-31:
Hi, they do work, but you have to not use the ArduinoHttsClient, you need to make the post and all the things by hand. In order to get it to work you need this (the server is the server and the port is 443):

```
TinyGsmClientSecure secureClient(modem, 1);
  Serial.println("Connecting to " + String(server));
  if (!secureClient.connect(server, port)) {
    Serial.println("... failed");
    restart_modem();
  } else {
    // Make a HTTP GET request:
    //secureClient.print(String("GET ") + "/alo.json" + " HTTP/1.1\r\n");
    //secureClient.print(String("Host: ") + server + "\r\n");
    //secureClient.print("Connection: keep-alive\r\n\r\n");

    // Make a HTTP POST request:
    String PostData = "{\"FLUJO\": \"555\", \"TEMP\": \"111\"}\r\n";
    secureClient.print(String("POST ") + "/" + imei + ".json" + " HTTP/1.1\r\n");
    secureClient.print(String("Host: ") + server + "\r\n");
    secureClient.print("Connection: keep-alive\r\n");
    secureClient.print("Content-Type: application/json\r\n");
    secureClient.print("Content-Length: ");
    secureClient.print(PostData.length());
    secureClient.print("\r\n\r\n");
    secureClient.print(PostData);

    // Wait for data to arrive
    uint32_t startS = millis();
    while (secureClient.connected() && !secureClient.available() &&
           millis() - startS < 30000L) {
      delay(100);
    };

    // Read data
    startS = millis();
    while (secureClient.connected() && millis() - startS < 5000L) {
      while (secureClient.available()) {
        Serial.write(secureClient.read());
        startS = millis();
      }
    }
    Serial.println();
  }
  secureClient.stop();
```

Hope it helps you out!

grichfitz, 2020-09-24:
Hya,
Has anyone managed to get HTTPS GET or POST to actually work?

I have tried many different approaches, surf´d the web and no-one seems to have this working. I am able to POST with HTTPS via wifi (with the wifi.h library, not TinyGsmClient.h library), but everyway I try fails with GSM on the T-Call.

I am aware there are issues with the older SIM800, but mine is, R14.18.

For example, I have tried mihai-lazar´s code above, but this fails on !secureClient.connect(server, port), with the following:
22:51:24.565 -> GPRS connected
22:51:24.565 -> Connecting to ********.com
22:51:24.565 -> AT+CIPCLOSE=1,1
22:51:24.565 -> 
22:51:24.565 -> ERROR
22:51:24.565 -> AT+CIPSSL=1
22:51:24.565 -> 
22:51:24.565 -> OK
22:51:24.565 -> AT+CIPSTART=1,"TCP","*******.com",80
22:51:24.565 -> 
22:51:24.565 -> OK
22:51:25.737 -> 
22:51:25.737 -> SMS Ready
22:51:26.574 -> 
22:51:26.574 -> 1, CONNECT FAIL

Please excuse the ******.com just not advertising my site. mihai-lazar does mention using port 443, but this makes no difference and anyway, via wifi port 80 works fine.

Anyone have any ideas? Example code? This would be extremely appreciated. And I know there are others needing help with this!

Here is my code: (again I´ve **** out personal data).

//==========HTTPS Client testing code ==== 

// Define how you're planning to connect to the internet
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false

//Connection Settings ---------------------------------------------------------------------------------------------------

//APN Details
const char apn[]      = "data.lycamobile.es"; // APN internet
const char gprsUser[] = "lmes"; // GPRS User
const char gprsPass[] = "plus"; // GPRS Password
#define GSM_PIN ""

//Wifi Details
const char wifiSSID[]  = "***************";
const char wifiPass[] = "***************";

// Server details
const char server[]   = "***************.com";
const char gopher[]   = "/get-server.php";
const char postman[]  = "/post-server.php";
const int  port       = 80;

String api_key = "hsgerFAgh3FT767";

//Hardware Settings ---------------------------------------------------------------------------------------------------

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

//Modem and serial configuration  -------------------------------------------------------------------------------------
#define TINY_GSM_MODEM_SIM800
#define TINY_GSM_RX_BUFFER 1024
[... 155 lines cut ...]
        Serial.write(secureClient.read());
        startS = millis();
      }
    }
    Serial.println();
  }
  
// Shutdown -----------------------------------------------------------------------------------------------------------
  secureClient.stop();
  SerialMon.println(F("Server disconnected"));

  #if TINY_GSM_USE_WIFI
      WiFi.disconnect();
      SerialMon.println(F("WiFi disconnected"));
  #endif
  #if TINY_GSM_USE_GPRS
      modem.gprsDisconnect();
      SerialMon.println(F("GPRS disconnected"));
  #endif

  // Do nothing forevermore
  while (true) {
    delay(1000);
  }
}

mihai-lazar, 2020-09-24:
mm i do not know what would the problem be, I use the same base I posted before and it works for Google Firebase (Firebase ONLY accepts https calls, and I had to figure out by my self how to do it and how to use the auth token also, that is the way I make the call to send the json with info). Try another https server and see if it works, maybe your server is having problems? Does it work with normal http? if not maybe the sim card has some problems accessing internet over gsm (I have that problem and had to change sim card, it was connecting to gsm but could not connect to any server, http or https).

Eiyvor, 2020-10-22:
You have to first upload your SSL certificate to SIM800 using the example sketch in tinyGsm repository called SIM800_SslSetCert, then use the TinyGsmClientSecure class in your sketch to start a secure connection(ie, https) on port 443.

mosix76, 2021-01-02:
Hello guys,
i'd like to send and receive message to my Telegram bot through https api and GPRS connection. I'm not so expert, I tried to understand the code you've sent, but it is not so simple, I don't know how to mount the pieces. Sorry for the request but, would be possible to have a complete working sketch to establish a HTTPS connection?

grichfitz, 2021-01-02:
I’ve been having issues getting HTTPS working to sent POST to my server. 
soz I can’t help, but defo following if someone has an answer!

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

ibrcic, 2021-07-31:
I am also unable to make HTTPS request, there is conflicting information on the web about the need of certificate installation. @mihai-lazar did you install any SSL certificates for connecting to the firebase?

attahiruj, 2022-02-11:
> Hi, they do work, but you have to not use the ArduinoHttsClient, you need to make the post and all the things by hand. In order to get it to work you need this (the server is the server and the port [...]

Please can you kindly provide a full sample code for posting to firebase? I'm having challenges with it. thanks

sam125, 2022-03-11:
> > I am also unable to make HTTPS request, there is conflicting information on the web about the need of certificate installation. @mihai-lazar did you install any SSL certificates for connecting to the firebase?
> 
> No I did not install any SSL, here is the full code @attahiruj and its still working on over 20 t-call
> 
> [SIM800L_POST_V4_test.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/8062948/SIM800L_POST_V4_test.zip)

This code is working with firebase?

soniritesh124, 2022-04-19:
Not working with HTTPS POST help for same...

ilgarbenli, 2023-12-05:
I couldn't connect googleapi for Geolocaton API, it uses port 443. Is there any solution for https connection.

> You have to first upload your SSL certificate to SIM800 using the example sketch in tinyGsm repository called SIM800_SslSetCert, then use the TinyGsmClientSecure class in your sketch to start a secure connection(ie, https) on port 443.

@Eiyvor I had googleapi cert folder, how can I implement this in my code, I looked SIM800_SslSetCert.ino but couldnt got how I implement this cert.