Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #72: TTGO-TCall sim800l net light

vitalstates, 2020-06-16:
I'm using TinyGSM library with client to upload data to php script on webserver. All works fine and I upload every 45 seconds or thereabouts so network is always connected but I establish a client connection with every upload.

sim800l documentation says net light should flash every 3 seconds when network is connected.....

mine flashes about 3 times every second.

Anybody got any ideas? and might it be possible to control the net light with tinygsm? I've waded through the source but haven't found anything yet.

lewisxhe, 2020-06-17:
Flashes three times per second, indicating that communication is in progress
Blinks once every three seconds, indicating a registered connection, but no communication
NETLIGHT cannot be controlled
[image]

peteakalad, 2020-06-17:
If it cant be disabled/controlled, has anyone physically removed/de soldered the led? 

Wondering if it makes much difference to current usage for running on batteries.

lewisxhe, 2020-06-18:
I was wrong, I can control the NET Light switch, and I can control the cycle, please check the manual.
5.2.14 AT+CNETLIGHT Close the Net Light or Open It to Shining
5.2.15 AT+CSGS Netlight Indication of GPRS Status
5.2.13 AT+SLEDS Set the Timer Period of Net Light

peteakalad, 2020-06-18:
Thanks lewisxhe.

For TinyGSM below works for me.

I've added call to turnOffNetlight() after modem.init() or modem.restart().  The first time its run it does flash maybe twice, but after that its off permanently, even after a reboot.

void turnOffNetlight()
{
  SerialMon.println("Turning off SIM800 Red LED...");
  modem.sendAT("+CNETLIGHT=0");
  modem.waitResponse();
  modem.sendAT("+CSGS=0");
  modem.waitResponse();
  modem.sendAT("&W");
  modem.waitResponse();
}

void turnOnNetlight()
{
  SerialMon.println("Turning on SIM800 Red LED...");
  modem.sendAT("+CNETLIGHT=1");
  modem.waitResponse();
  modem.sendAT("+CSGS=1");
  modem.waitResponse();
  modem.sendAT("&W");
  modem.waitResponse();
}

vitalstates, 2020-06-19:
thanks Peteakalad
I was wondering what was the quickest way to send AT commands using tinyGSM. The way it flashes so quickly seems to bring to mind an error situation.

That precisely answers my original post.

lewisxhe, 2020-06-22:
It seems that you have solved it, please reopen if there are still problems.

closed 2020-06-22 by lewisxhe