Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #74: SD card Issue

jenil163, 2020-06-22:
I Try to store data in SD CARD  WITH TTGo but hear all SPI uses some different connection like 23,5 are used for  SIM800L POWER AND RST SECOND SPI is For 13 for built-in led and third for FSPI so how can I connect and exchange data with SD card I see Issue #17  but not working with TTGO-T-CALL-SIM800L?
Anyone experience Please Help

lewisxhe, 2020-06-23:
You can only use unused pins ，can you tell which pins you use?

jenil163, 2020-06-23:
I use   clk=14, mosi=15, miso=2, cs=13 not working after connecting Mosi pin Module is every time reset

jenil163, 2020-06-23:
[image]

I use this device

lewisxhe, 2020-06-24:
You used these pins, did you initialize it? As described here 

 #17

jenil163, 2020-06-24:
I try But the same problem so initialization failed module works fine I check-in  Nodemcu I send you Code

jenil163, 2020-06-24:
When I use  #17 pin  then this type of error I See
[image]
[image]

Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
An error occurred while uploading the sketch

lewisxhe, 2020-06-24:
In T-Call, you can directly use SPI instead of HSPI, like this
```
    #include <SPI.h>
    #include <SD.h>
    SPI.begin(sck, miso, mosi);
    if (!SD.begin(sd_cs)) {
        // failed
    } else {
        //pass
    }
```
There is no error in your code, but did you upload it with the SD card inserted? GPIO12 is used as the flash voltage selection when ESP32 starts, please change a pin if you can

jenil163, 2020-06-29:
First I try with inbuilt SPI pin no 23,5,18,19 but it's not working, TTGO support 8gb memory card?

lewisxhe, 2020-06-30:
I am updating the warehouse and I will add an sd example.

lewisxhe, 2020-07-01:
The code and instructions have been updated, I can use each version! Please check!

natanel80, 2020-07-04:
@lewisxhe hi can you help me please i still not figure which pins i need to connect in order to work with sim800l and external sd together or if you have any code i will more than happy to see it

jenil163, 2020-07-06:
@natanel80 which Device you use?  same or other

lewisxhe, 2020-07-06:
@natanel80 I have uploaded the code, and the picture connected to the SD, have you checked it?

natanel80, 2020-07-06:
@lewisxhe yes its all the time say "Card Mount Failed"

lewisxhe, 2020-07-06:
You need to check your connection, do you apply the same SD card module as me? The SD card needs to be connected to 5V to work.

natanel80, 2020-07-06:
this is the code right 
[image]

lewisxhe, 2020-07-06:
This is correct, I have verified it, and I have taken pictures and put them in the README.

natanel80, 2020-07-06:
[image]

natanel80, 2020-07-06:
@lewisxhe the sd its ok?

lewisxhe, 2020-07-06:
I’m right, please change the cable and try again. If it still doesn't work, please check the module for problems.

natanel80, 2020-07-06:
@lewisxhe  ok thank you

Cutiino, 2020-07-17:
Hello, I also am trying work with SD card and have a trouble with the 5V power. When I connect the SD card, the sim don't initialize but when I disconnect the SD, the sim initialize without problem.

closed 2020-09-18 by lewisxhe

uvce12, 2022-02-23:
If you are using VSPI and the GPRS, pin 23 is used for powering up the SIM800L.

If you using HSPI, the pin 12 is always pulled high.

I was running my E-Ink on SPI and had no problems until I wrote the code for GPRS. Then, I redefined MOSI on VSPI to pin 21 instead of pin 23 and my SPI device works just fine.