Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #79: LilyGo-T-Call-SIM800L - How to detect power source USB-C or BATTERY

Vitaminazeta, 2020-07-26:
Hello everyone,
I'm looking to a way to detect if my LilyGo-T-Call-SIM800L is powered from USB-C connector or if running from battery.
My first option was to add an extra power supply (220VAC to 5V) connected only to a GPIO pin and detect if there is power. I wish to avoid this solution so save a second power suply unit on the project.
My second option was to connect the 5Vdc coming from USB-C to one of the GPIO pin. The problem here is how to get USB-C 5v. USB-C connector has a very small pins and I wan't be able to solder a wire without destroy the whole PCB.

Anyone knows if is possbile to detect power source on LilyGo-T-Call-SIM800L.
Actually I'm using Adafruits_FONA library where I should be able to measure battery capacity but not power source.
many thanks

peteakalad, 2020-07-26:
If you have an AXP192 model some of the methods from the AXP library may help see:
https://github.com/lewisxhe/AXP202X_Library/blob/master/src/axp20x.h

lewisxhe, 2021-01-19:
Issues have not been active for a long time and will be closed. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

streem-ua, 2023-01-06:
Hey, @Vitaminazeta , have you solved your issue? I'm in such situation, like yours, trying to detect power source. 
Thanks!

Vitaminazeta, 2023-01-15:
Hello!
I give up After months. If you find a way please be so kind to let me know

Giacomo

streem-ua, 2023-01-15:
@Vitaminazeta Hey, go here! I managed how to know USB/VBAT
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/222

lewisxhe, 2023-01-15:
see there https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/222

Vitaminazeta, 2023-01-15:
> see there #222

Many thanks lewisxhe