Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #80: Unable to Connect to GSM When Battery Below 3.9V

peteakalad, 2020-07-26:
I have a newer AX192 SIM800L model t-call using a 3.7V 8000mah lithium polymer battery via the JST connector.

I fully charge the battery with USB-C connected, and battery voltage is just under 4.2V as expected.  

I disconnect the USB to test performance on battery alone.

I have a tinygsm sketch similar to the example which connects the modem, makes 1 HTTP request and then deep sleep for 60seconds, this manages 43 boots before it fails to init the modem.

It seems that once the battery voltage is below 3.9V it is unable to connect to the GSM network, I see the status light go on when initializing the modem then it goes shortly afterwards.

I'm assuming there is insufficient current to connect/transmit, but I would have thought that at 3.9V the battery should be around 40% full.

I have tried connecting a 470uF capacitor across the battery, but this does not help.

Any other suggestions to make full use of the charge in the battery?

peteakalad, 2020-08-09:
Put it down to a bad battery, closing.

peteakalad, 2020-08-10:
New batteries 4 x 18650 cells in parallel, same problem, reopening.

I debugged some of the voltages below, I note a low sysIPSOutVolt, perhaps its this causing the issue.

When the problem is arising (<= 3.9V) it boots and gets to waiting for network.  After a short period the status light goes out and the modem stops responding.   A multimeter confirms the batteries are at 3.9V with no load, but I have 4 cells and there ought to be sufficient power/current for the modem to connect.  As above I am suspicious of the low IPS voltage from the AXP192, I believe this powers the SIM800L?

I have tried:
- disabling the battery check in the modem via modem.sendAT(GF("+CBATCHK=0"));
- setting to 1800Mhz only via modem.sendAT(GF("+CBAND=\"EGSM_DCS_MODE\""));

Battery Powered (fails at waiting for network)
=============================
isCharging 0
isVBUSPlug 0
vbusVoltage 0.00
vbusCurrent 0.00
battVolt 3897.30
battChargeCurrent 0.00
battDischargeCurrent 84.50
sysIPSOutVolt 2781.00
DCDC3Volt 3300.00

USB Powered (works ok)
================
vbusVoltage 5049.00
vbusCurrent 0.00
battVolt 0.00
battChargeCurrent 0.00
battDischargeCurrent 0.00
sysIPSOutVolt 3551.00
DCDC3Volt 3300.00

peteakalad, 2020-08-12:
I have run another test, running the board on a loop repeatively connecting the GSM network on battery only.  Again this refused to connect to GSM network closer to 3.8V this time.  

I took the 4 x 2200mAh (confirmed capacity) 18650 cell batteries in parallel off the T-Call and tested on a discharge tester with a load around 2.3Amps (same/similar to the t-call).  I was able to extract a further 6011mAh (of 8800mAh total) between 3.8V and 2.5V (load voltage, normal range for 3.7V cells).  

**This test demonstrates that _68%_ (!!) of the battery capacity is not able to be used by the board due it failing at around 3.8 / 3.9V.**   For a board intended to be run from battery this really is not good.

The SY8089 regulator fed by IPSout has a cut off at 2.7V according to the datasheet.  Given the AXP192 indicates a sysIPSOutVolt of 2.781 when the battery is around 3.9V,  under load this going to drop below that presumably causing the issue I'm seeing with insufficient power being provided for the SIM800L to connect to the GSM network.

It is unfortunate that it looks like this is a another revision of the board where the power management needs to be skipped :(  Or perhaps someone ( @lewisxhe ? ) may know how to setup the AXP192 so keeps the IPSOUT at voltage above 2.7V that the regulator requires even when the supply battery voltage is < 3.9V.   I have difficulty reading the chinese datasheet to confirm if there are any registers not setup in the AXP192 library which may help.

jsoromom, 2020-09-05:
Same issue here...
Rather than batteries (as they could be bad), i'm using a Power supply (DPS5005 ) and it the Voltage is down to 3.8V the board doesn't work. The board doesn't work properly...

peteakalad, 2020-09-05:
Thanks for verifying my findings, good idea to test with bench PSU.  Don’t have one unfortunately. 

Does it fail to GSM connect at lower voltages as I think that’s highest current draw? Guess yours was already connected in the video?

But anyway flawed design given its to be powered by a 3.7v lithium based battery with typical voltage range of 4.2v to 3.0v :(

jsoromom, 2020-09-05:
At 3.66v the board gets reset every just a few seconds failing to GSM connect doing looping. Not good...

rajeshatuce, 2020-09-08:
any possible solution to this problem? Can we directly power the board using 3.3V by connecting to 3v3 pin?

steckpro, 2020-10-09:
Take a 470uF capacitor between 5V an GND and the problem is solved

peteakalad, 2020-11-16:
@steckpro I added a 470uF cap across the GND and 5V points at the side of the board - same issue, unable to connect once the batteries reach around 3.9V.   Are you connecting the cap in the same way or somewhere else on the board?  You have a AX192 based board?

steckpro, 2020-11-16:
I bought this item: 
https://a.aliexpress.com/_mMY20Wv

My module connects by battery under 3.0 V without problems, but without the capacitor, the T-Call ist restarting continously. I don't know, the T-Call 1.3 is the older version?

Maybe the code is the Problem?

I connected the capacitor just between 5V and GND on GPIO

jsoromom, 2020-11-16:
Hi @steckpro, your board is managed by the IP5306 rather than the AX192. I've got both boards at home so I'll do the test and come back to you. I don't really understand the link between that capacitor between 5V and GND and the use of Batteries but, for sure, I'll give it a try...
Thanks for your comments.

jsoromom, 2020-12-02:
Test done, the result is the same when a capacitor is placed between 5V and GND. It has sense according my last comment (the 5V power pin on the board is the input)
Anyway, I added a Small Board LM2587 (fixed to 4.15V) between the battery and the battery connector. At the same time I'm taking measurements using an analog input for the Battery (voltage divider included). The result is quite good after first trials. At the moment (test in progress) battery is 3.13V and the board is still working properly.

dtheodor79, 2020-12-18:
@jsoromom thanks for the update. Unfortunately I also have the exact same problem. My question is that if we insert the LM2587 between the battery and its connector, can we charge it via the ttgo USB port?

Thanks,
dtheodor

steckpro, 2020-12-18:
I don’t know, why you have these problems. I have made a test by connecting every 5 minutes to network and post the battery charge to MySQL DB, here the picture
[image]

I think, your code is not Right, can you Post your Code? Do you set power boost while transmitting?

peteakalad, 2020-12-18:
@steckpro there is no power boost with the AXP192.  I am using the same tinyGSM provided in the sample extended to loop forever.  

The reason for the difference is because **you have a different board**

Based on your aliexpress link you have the old board, which has a different power management chip and uses IP5306 vs the newer AXP192 which we are discussing.

steckpro, 2020-12-18:
Oh okay, now I understand, then I can't help you at all. Do you have a link to buy the board?

peteakalad, 2020-12-18:
This one is uses the AXP192 power management:
https://www.aliexpress.com/item/4000959701330.html

You can confirm the difference by the date printed on the board, see the table on the top page of this repo: https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800#product-

jsoromom, 2020-12-18:
Anyway, @steckpro,it would be brilliant if you could share your working code (for IP5306 board). Please, do it if you are Ok with that.

Thanks in advance

lewisxhe, 2021-01-19:
Issues will be closed if they have not been active for a long time. If there is a problem, please reopen it.

closed 2021-01-19 by lewisxhe

jsoromom, 2021-01-19:
Any solution then?

lewisxhe, 2021-01-19:
@jsoromom Do you have any questions? You can create a new issuse

jsoromom, 2021-01-19:
Sure...according specs board can work from 2,7 V-3,6 V.
How can we use that board according those numbers feeding from battery? 

Not a new issue, this is the same one as OP.

Thanks

lewisxhe, 2021-01-19:
This is wrong. The details are about the power supply of the esp32 chip, not the board. The board should be at least 4.2V. Because 800x uses 4.0V voltage, although it has a boosted DC, lower than the battery voltage will still make the modem work abnormally.@jsoromom

peteakalad, 2021-01-19:
But its sold to be powered by batteries which will not always have a 4.0V!!

As above when voltage is less than 3.8 / 3.9 V on my set of parallel 18650
cells the modem no longer connects.

The standard lion discharge curve illustrates how this leaves significant
unused capacity in the batteries which is poor (68% of the battery capacity
was not used from my test!)
https://batteryuniversity.com/learn/article/discharge_characteristics_li

This is not resolved?

lewisxhe, 2021-01-19:
@peteakalad
 At present, there is no good solution, only large-capacity lithium batteries or 18650 battery packs can be used.

jsoromom, 2021-01-19:
Not good news at all... Battery connector will be 95% useless according your comments, I'm afraid.

ClemensGruber, 2021-07-02:
I could use the SIM800C with AXP as power management chip down to ~ 3.7 V: 

Datum/Zeit,Gewicht,Temperatur,Feuchte,Luftdruck,Batterie-Spannung
2021/06/29 23:51:25,  -0.006, 21.6, 67.4, 1008.3, 3.68
2021/06/29 23:56:39,  -0.007, 21.6, 67.1, 1008.3, 3.68

You may check your cables and also battery contacts, see https://community.hiveeyes.org/t/inbetriebnahme-der-datenubertragung-per-mqtt-und-tinygsm-bibliothek-uber-sim800-modul-auf-esp8266/849/17

ClemensGruber, 2022-01-04:
Unfortunately I have now the problem with a new batch of boards! See issues #189, the older boards are working down to 3.7 V but not the new one! :-( All boards -- working and not working are SIM800_**C**_ T-Calls and have the same silkscreen lable on the T-Call PCB but other ESP32 modules.

jsoromom, 2022-01-04:
As summarize... the 4 boards I've got are inside a drawer... I'm using right now the combo Lolin D32 + SIM800L. It works as expected.
[image]

ClemensGruber, 2022-01-05:
Thanks for your recommendation @jsoromom. I have spotted the DFRobot FireBeetle as it has a deep sleep current of 10 uA only. Do you have numbers for your Lolin D32 measured or is deep sleep not relvant for your application? Did you add some capacitor in the SIM800 power line or does it work without additional caps? The SIM800 pins are not 3V3 tolerant, so I have planed a voltage divider in case the T-Call is a never ending disappointing story. I have one (!) T-Call working down to 3.56 V see https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/189#issuecomment-1005485891 but unfortunately no feedback form @LilyGO nither here nor via the tech support email, disappointing!

ClemensGruber, 2022-01-05:
> It is unfortunate that it looks like this is a another revision of the board where the power management needs to be skipped :( Or perhaps someone ( @lewisxhe ? ) may know how to setup the AXP192 so [...]

Can someone from Lilygo provide more information, e.g. @lewisxhe ? There are working boards outside. So it would be interesting if a hardware change was done or if e.g. some default settings on the AXP can make the difference. Both information would be worthfull to me at least to give it up in case there is a hardware bug or fix it in case software can do the job. Documentation about the AXP is rare, only this most in chinese written documentaion, and I do not know any chinese, sorry!