Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #82: Add example to retrieve the network time

rgl, 2020-07-29:
It would be useful to have an example on how to get the network time. It seems it can de done with:

https://github.com/vshymanskyy/TinyGSM/blob/cddd356d0cca698ffcd324c0ca4702c1b92a2142/src/TinyGsmClientSIM800.h#L46

Can we have an example of that?

peteakalad, 2020-07-29:
Did you look at the tinyGSM examples?

E.g. https://github.com/vshymanskyy/TinyGSM/blob/cddd356d0cca698ffcd324c0ca4702c1b92a2142/examples/AllFunctions/AllFunctions.ino#L404

rgl, 2020-07-30:
I did, I thought this was a normal thing to do when we use this module that it would make sense to pull that example to this repository. But, I'm not sure anymore. So I'll close this. Sorry for the trouble.

closed 2020-07-30 by rgl