Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #87: IP5306 instability issues with SIM800L_IP5306_VERSION_20190610

mbobinger, 2020-08-12:
Dear TTGO-fellows,

I'm having great fun with this product version:
SIM800L_IP5306_VERSION_20190610

My code works fine on it after implementing a few fixes:
1) 470uF cap soldered between GND and 5V for SIM800 current spikes
2) 2.0mm pitch JST plugs replaced by thicker cables for the LiPo battery
3) calling setupPMU in the setup to keep the battery charger powered, the exact function is explained in utilities.h

However: I'm still getting some instability issues after several transmissions and that seems really hard for me to troubleshoot now. I've observed that the module can hang in some zombie mode that consumes around 50mA, when the voltage drops below a critical value.

I've seen a similar issue opened in this thread. Is there by now some kind of a best practice how to use the TTGO module in battery powered mode?
What is the latest stable Hardware version that you recommend and could you please send me a link to that?

Thank you!

lewisxhe, 2020-09-03:
Already replied in another thread this will be closed

closed 2020-09-03 by lewisxhe

Rob58329, 2024-04-05:
re. the LilyGo-T-Call-SIM800 (ESP32).

@mbobinger suggestion of adding a 470uF capacitor across the 5V pin and ground worked for me.

I note that the IP5305 is apparently only able to supply a maximum of 1.2A at 5V, and the SIM800 can draw pulses of upto 2A.  However when the IP5305 is powered from the USB 5V, it seems to be able to supply the SIM800 OK.  But if the IP5305 is powered from a 3.7V (to 4.2V) lipo battery, although the ESP32 will boot OK, as soon as the ESP32 turns on the power to the SIM800 module, the IP5305 can’t supply sufficient current (probably the current pulses) and the 5V line drops to 2V or less, which usually causes the ESP32 to crash, de-powering the SIM800, which then allows the 5V to rise again and the ESP32 re-starts.  And the boot-crash-boot-crash.. cycle repeats.

However adding the above 470uF capacitor on the 5V line seems to be able to supply the current peaks drawn by the SIM800 on turn-on, and so the 5V rail isn’t pulled down and the ESP32 continues to run.  (To boot the ESP32 on battery, you have to connect the battery, and then press the “reset” button which tells the IP5305 to turn the power ON.)

Interestingly, I note that (even with this 470uF capacitor) if the ESP32 is running on 5V USB Power, and you remove the USB Power, the IP5305 does NOT (USUALLY) instantly swap to using the 3.7V battery.  Instead the IP5305 allows it’a 5V output to drop to the point where the ESP32 memory starts to fail, which causes an ESP32 crash and re-boot, but by this time the IP5305 has brought the 5V back up, and the ESP32 will re-boot ok.

So, as long as you have programmed the ESP32 to recover from this crash, this is OK and can be got around.

NB.  Although the above “crash on USB-power-failure” happens about 90% of the time, about 1 time in 10 the IP5305 appears to succeed in switching over to the 3.7V battery fast enough that the ESP32 doesn’t notice, BUT this is not very often.

kaly7, 2025-01-06:
> re. the LilyGo-T-Call-SIM800 (ESP32 with IP5305 power controller).
> 
> @mbobinger suggestion of adding a 470uF capacitor across the 5V pin and ground worked for me.
> 
> I note that the IP5305 is apparently only able to supply a maximum of 1.2A at 5V, and the SIM800 can draw pulses of upto 2A. However when the IP5305 is powered from the USB 5V, it seems to be able to supply the SIM800 OK. But if the IP5305 is powered from a 3.7V (to 4.2V) lipo battery, although the ESP32 will boot OK, as soon as the ESP32 turns on the power to the SIM800 module, the IP5305 can’t supply sufficient current (probably the current pulses) and the 5V line drops to 2V or less, which usually causes the ESP32 to crash, de-powering the SIM800, which then allows the 5V to rise again and the ESP32 re-starts. And the boot-crash-boot-crash.. cycle repeats.
> 
> However adding the above 470uF capacitor on the 5V line seems to be able to supply the current peaks drawn by the SIM800 on turn-on, and so even when running on battery the 5V rail isn’t pulled down and the ESP32 continues to run. (To boot the ESP32 on battery, you have to connect the battery, and then press the “reset” button which tells the IP5305 to turn the power ON.)
> 
> Interestingly, I note that (even with this 470uF capacitor) if the ESP32 is running on 5V USB Power, and you remove the USB Power, the IP5305 does NOT (USUALLY) instantly swap to using the 3.7V battery. Instead the IP5305 allows it’a 5V output to drop to the point where the ESP32 memory starts to fail, which causes an ESP32 crash and re-boot, but by this time the IP5305 has brought the 5V back up using the battery, and the ESP32 will re-boot ok.
> 
> So, as long as you have programmed the ESP32 to recover from this crash, this is OK and can be got around.
> 
> NB. Although the above “crash on USB-power-failure” happens about 90% of the time, about 1 time in 10 the IP5305 appears to succeed in switching over to the 3.7V battery fast enough that the ESP32 doesn’t notice, BUT this is not very often.

It looks like I'm having this problem too. The 470uF capacitor should be placed between GND and 5V, as written, but where is it on the motherboard? Did you mean the 5V pin here, the one above SPK+?

Rob58329, 2025-01-06:
@kaly7:  I didn't put the 470uF capacitor on the "LilyGo-T-Call-SIM800" board itself, but instead on the 5V and ground pins on the 0.1" board connector (ie, the 5V supply as near as possible to the 5V pin of the "LilyGo-T-Call-SIM800").  NB. You can still power the "LilyGo-T-Call-SIM800" through it's USB-C connector.

kaly7, 2025-01-06:
@Rob58329 : Thank you for your quick, correct answer.