Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #89: Flash failed: ESP32: Timed out waiting for packet header

mariusgrams, 2020-08-16:
Hi guys, I have the problem that I can't upload software to the board. I am using Arduino IDE and selected ESP32 Dev Module as Board. Uploading new software has worked before but rarely. I have already read the other issue but none of it worked. I tried to connect pin 0 with pin GND but it did not work 

[image]

I always get the error: Failed to connect to ESP32: Timed out waiting for packet header.
Could someone explain in more detail how they fixed the problem?

_Originally posted by @mariusgrams in https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/31#issuecomment-660025749_

esox01, 2020-08-28:
Hello, my solution was to connect the board via USB 2 (USB 3 was not working). No reset button required.

closed 2020-09-03 by lewisxhe

mariusgrams, 2020-09-17:
Hey, when I follow your instructions, I get the following message in my serial monitor:

ets Jun  8 2016 00:22:57

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
```
I think my board is not in download mode right? How do I put my board into download mode?