Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #91: Communication Issue

GDXFR, 2020-09-03:
Hi guys,

I'm a new user with a SIM800L_IP5306_VERSION_20190610, I experienced communication issue with the SIM800L, it doesn't seem to receive or respond to serial com, I'm testing with https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_TinyGSM example:
- turnOffNetlight() has not effect
- modem.getModemInfo() does not return anything
- modem.waitForNetwork(240000L) keeps sending "AT+CREG?"

Battery is connected (4.2V measured).
Debug with USB.
Network seems catch as red light slow down after a few seconds while SIM card is inserted.

Any help would be most welcome.

GDXFR, 2020-09-03:
Bought on Ali here : https://www.aliexpress.com/item/4000334093712.html?spm=a2g0s.9042311.0.0.27424c4dS31jw2

GDXFR, 2020-09-03:
I experimented with this code to test Serial communications:

```
void setupModem()
{
    // Keep reset high
    pinMode(5, OUTPUT);
    digitalWrite(5, HIGH);

    pinMode(4, OUTPUT);
    pinMode(23, OUTPUT);

    // Turn on the Modem power first
    digitalWrite(23, HIGH);

    // Pull down PWRKEY for more than 1 second according to manual requirements
    digitalWrite(4, HIGH);
    delay(100);
    digitalWrite(4, LOW);
    delay(1000);
    digitalWrite(4, HIGH);
}

void setup() {
  setupModem();
  Serial.begin(9600);
  Serial1.begin(9600, SERIAL_8N1, 26, 27);
}

void loop() { // run over and over
 if (Serial1.available()) {
 Serial.write(Serial1.read());
 }
 if (Serial.available()) {
 Serial1.write(Serial.read());
 }
}
```

IO27 seems to receive what I transmit from RealTerm (I check that with a multimeter: drop from 3.3V to lower value when a message is sent), however no change on IO26 which is stable at 2.8V (strange?).

GDXFR, 2020-09-13:
Hello, shall I assume it is dead?

GDXFR, 2020-09-14:
Received SIM800C_AXP192_VERSION_20200609 today, working fine.. So just lost 15€ and lot of time trying to resolve this issue.

lewisxhe, 2020-09-18:
I think you seem to have solved this problem and the problem will be closed

closed 2020-09-18 by lewisxhe

GDXFR, 2020-09-18:
Not really solved as I had to buy a new one..