Xinyuan-LilyGO/LilyGo-T-Call-SIM800 #95: Couldn't find GSM SIM800L

thalysmg, 2020-10-09:
I'm trying to upload a simple code where I send and receive an SMS. But the SIM800L won't be recognized for nothing. The red led, near the SIM800L will never turn on.
I would be glad if anyone could help me.

Right below is my code:

```
// WifiManager
#include <WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

// ESP32 + GSM MODULE
#include "Adafruit_FONA.h"
#define SIM800L_RX     27
#define SIM800L_TX     26
#define SIM800L_PWRKEY 4
#define SIM800L_RST    5
#define SIM800L_POWER  23
char replybuffer[255];
HardwareSerial *sim800lSerial = &Serial1;
Adafruit_FONA sim800l = Adafruit_FONA(SIM800L_PWRKEY);
uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);
#define LED_BLUE  13
#define RELAY 14
String smsString = "";

void setup() {
  //Setup Receiver SMS
  pinMode(LED_BLUE, OUTPUT);
  pinMode(RELAY, OUTPUT);
  pinMode(SIM800L_POWER, OUTPUT);
  digitalWrite(LED_BLUE, HIGH);
  digitalWrite(SIM800L_POWER, HIGH);
  Serial.begin(115200);
  Serial.println(F("ESP32 with GSM SIM800L"));
  Serial.println(F("Initializing....(May take more than 10 seconds)"));
  delay(10000);
  // Make it slow so its easy to read!
  sim800lSerial->begin(4800, SERIAL_8N1, SIM800L_TX, SIM800L_RX);
  if (!sim800l.begin(*sim800lSerial)) {
    Serial.println(F("Couldn't find GSM SIM800L"));
    while (1);
  }
  Serial.println(F("GSM SIM800L is OK"));
  char imei[16] = {0}; // MUST use a 16 character buffer for IMEI!
  uint8_t imeiLen = sim800l.getIMEI(imei);
  if (imeiLen > 0) {
    Serial.print("SIM card IMEI: "); Serial.println(imei);
  }
  // Set up the FONA to send a +CMTI notification
  // when an SMS is received
  sim800lSerial->print("AT+CNMI=2,1\r\n");
  Serial.println("GSM SIM800L Ready");
  
  //Setup SEND EMAIL
  WiFiManager wifiManager;
  //cria uma rede de nome ESP_AP com senha 12345678
  wifiManager.autoConnect("ESP_32", "00000000");
}

// More info receiver SMS Between setup and loop
long prevMillis = 0;
int interval = 1000;
char sim800lNotificationBuffer[64];          //for notifications from the FONA
char smsBuffer[250];
boolean ledState = false;

void loop() {
  // JUST SHOW SMS
  if (millis() - prevMillis > interval) {
    ledState = !ledState;
    digitalWrite(LED_BLUE, ledState);
    prevMillis = millis();
  }
  char* bufPtr = sim800lNotificationBuffer;    //handy buffer pointer
  if (sim800l.available()) {
    int slot = 0; // this will be the slot number of the SMS
    int charCount = 0;
    // Read the notification into fonaInBuffer
    do {
      *bufPtr = sim800l.read();
      Serial.write(*bufPtr);
      delay(1);
    } while ((*bufPtr++ != '\n') && (sim800l.available()) && (++charCount < (sizeof(sim800lNotificationBuffer) - 1)));
    //Add a terminal NULL to the notification string
    *bufPtr = 0;
    //Scan the notification string for an SMS received notification.
    //  If it's an SMS message, we'll get the slot number in 'slot'
    if (1 == sscanf(sim800lNotificationBuffer, "+CMTI: \"SM\",%d", &slot)) {
      Serial.print("slot: "); Serial.println(slot);

      char callerIDbuffer[32];  //we'll store the SMS sender number in here

      // Retrieve SMS sender address/phone number.
      if (!sim800l.getSMSSender(slot, callerIDbuffer, 31)) {
        Serial.println("Didn't find SMS message in slot!");
      }
      Serial.print(F("FROM: ")); Serial.println(callerIDbuffer);

      // Retrieve SMS value.
      uint16_t smslen;
      // Pass in buffer and max len!
      if (sim800l.readSMS(slot, smsBuffer, 250, &smslen)) {
        // Atribui a mensagem a baixo em smsString
        smsString = String(smsBuffer); 
        SendSMS();
      }
    }
  }
}

void SendSMS() {
  Serial.println("Sending SMS...");               //Show this message on serial monitor
  sim800l.print("AT+CMGF=1\r");                   //Set the module to SMS mode
  delay(100);
  sim800l.print("AT+CMGS=\"+5583999701895\"\r");  //Your phone number don't forget to include your country code, example +212123456789"
  delay(500);
  sim800l.print("Olá Thalys!");       //This is the text to send to the phone number, don't make it too long or you have to modify the SoftwareSerial buffer
  delay(500);
  sim800l.print((char)26);// (required according to the datasheet)
  delay(500);
  sim800l.println();
  Serial.println("Text Sent.");
  delay(500);
}
```

lewisxhe, 2020-10-10:
You can use the example provided by LilyGo for testing, I have not used Adafruit_FONA.

toph84, 2020-12-05:
I've been successful with Adafruit_FONA and adapted a very similar script for me as well to do some relay switching with SMS. Are you sure you have SIM800L-version? If it's a SIM800C you have to use PIN 25 for modem power.

SIM800C_AXP192
MODEM POWER  25

SIM800L_AXP192/SIM800L_IP5306 (V1.3 & V1.4)
MODEM POWER   23

Best wishes
Chris

lewisxhe, 2021-01-28:
As @toph84 said, you must turn on the power to work.
The Adafruit_FONA function is too simple, so I did not test it. Recommend you to use TinyGSM

closed 2021-01-28 by lewisxhe